dji-sdk/Payload-SDK #2: 大疆无人机开发者问题

ChrisChen365, 2019-05-07:
This Github issue is synchronized with Zendesk:

**Ticket ID:** [#23563](https://djisdksupport.zendesk.com/agent/tickets/23563)
**Requester:** doc.wangjingui@foxmail.com
**Issue escalated by:** 平安
**premium support code:** 
**Language:** 
**Industry and Interest:** 
**Product:** 
**Platform:** 
**Main Category:** 
**Sub-Category:** 
**SDK Category:** 
**Ticket reopened(DON'T click):** 
**Difficulty Type:** 

**Original ticket description:**

<blockquote><div class="zd-comment" dir="auto">
<p dir="auto">飞机型号：mavic 2
<br>SDK版本：4.9
<br>android版本：8.0
<br>你好，我运行媒体管理这个Demo，他无法获取飞机上SD卡照片，请问一般是什么问题，该如何解决？</p>

<p dir="auto">This email and any attachments thereto may contain private, confidential, and privileged material for the sole use of the intended recipient. Any review, copying, or distribution of this email (or any attachments thereto) by others is strictly prohibited. If you are not the intended recipient, please contact the sender immediately and permanently delete the original and any copies of this email and any attachments thereto.</p>

<p dir="auto">此电子邮件及附件所包含内容具有机密性，且仅限于接收人使用。未经允许，禁止第三人阅读、复制或传播该电子邮件中的任何信息。如果您不属于以上电子邮件的目标接收者，请您立即通知发送人并删除原电子邮件及其相关的附件。</p>
</div></blockquote>

ChrisChen365, 2019-05-07:
test

ChrisChen365, 2019-05-07:
Zendesk ticket #[23550](https://djisdksupport.zendesk.com/agent/tickets/23550) has been linked to this issue.

ChrisChen365, 2019-05-07:
*Comment made from Zendesk by 平安 on 2019-05-07 at 09:23:*

<blockquote><div class="zd-comment" dir="auto">sadfasda<br><br>
</div></blockquote>

ChrisChen365, 2019-05-07:
test

ChrisChen365, 2019-05-07:
*Comment made from Zendesk by 平安 on 2019-05-07 at 09:33:*

<blockquote><div class="zd-comment" dir="auto">SADFASDF<br>
</div></blockquote>

ChrisChen365, 2019-05-07:
*Comment made from Zendesk by 平安 on 2019-05-07 at 09:35:*

<blockquote><div class="zd-comment" dir="auto">11112223333<br>
</div></blockquote>

pcler, 2019-05-07:
12345

pcler, 2019-05-07:
aaaaaaaaaaaaaaaaa

pcler, 2019-05-07:
ccccccccccccccccc

ChrisChen365, 2019-05-07:
*Comment made from Zendesk by 平安 on 2019-05-07 at 09:41:*

<blockquote><div class="zd-comment" dir="auto">ddddddddd<br>
</div></blockquote>

ChrisChen365, 2019-05-07:
*Comment made from Zendesk by Zoe.McAtee on 2019-05-07 at 09:44:*

<blockquote><div class="zd-comment" dir="auto">
<p dir="auto">-- This notification was sent from JIRA DS-213 to all linked Zendesk Support tickets by Chris Chen. --</p>
<p dir="auto">了解问题</p>
</div></blockquote>

ChrisChen365, 2019-05-07:
*Comment made from Zendesk by 平安 on 2019-05-07 at 09:50:*

<blockquote><div class="zd-comment" dir="auto">
<p dir="auto">-- 此通知由 平安 从 Zendesk Support 工单 <a href="/agent/tickets/23550" target="_blank" rel="ticket">#23550</a> 发送到所有相关 JIRA 问题。--</p>
<p dir="auto">赶紧啊</p>
</div></blockquote>

ChrisChen365, 2019-05-07:
*Comment made from Zendesk by 平安 on 2019-05-07 at 09:52:*

<blockquote><div class="zd-comment" dir="auto">
<p dir="auto">-- 此通知由 平安 从 Zendesk Support 工单 <a href="/agent/tickets/23550" target="_blank" rel="ticket">#23550</a> 发送到所有相关 JIRA 问题。--</p>
<p dir="auto">@chris.chen 速度</p>
</div></blockquote>

ChrisChen365, 2019-05-07:
*Comment made from Zendesk by Zoe.McAtee on 2019-05-07 at 09:55:*

<blockquote><div class="zd-comment" dir="auto">
<p dir="auto">-- This notification was sent from JIRA DS-213 to all linked Zendesk Support tickets by Chris Chen. --</p>
<p dir="auto">搞定了</p>
</div></blockquote>

pcler, 2019-05-07:
lallalallallalla

closed 2019-05-08 by ChrisChen365