dji-sdk/Payload-SDK #23: About DJI-H264 encoding for streaming

OpusK, 2022-01-10:
Hi, all.

I'm using the PSDK to develop Camera Streaming.
Existing streaming with H/W capture card encoder was working well.
And now I want to use CUDA encoder with other capture card because of the discontinuation issue.

The newly encoded image has passed [DJIPSDKVideoStreamCheckTool](https://github.com/dji-sdk/DJIPSDKVideoStreamCheckTool)'s test, but nothing is output from DJI App like below.
It comes out transparent. (Main screen is FPV Camera)
[image]

The test environment is as follows.
- PSDK : V3.0.0-beta.0-build.1440
- Aircraft : 03.00.0101
- Remote Controller : 03.01.0000

And below is additional information.
- [PSDK Log archive file](https://github.com/dji-sdk/Payload-SDK/files/7837927/DJI_0073_20220110_18-28-55.zip)
- [encoded media archive file](https://github.com/dji-sdk/Payload-SDK/files/7837965/dslr_rec_latest.zip)

I have confirmed that there is no problem with encoding with the following documents and tools, but if I have a part I missed, please let me know.
- [DJI-H264 spec](https://developer.dji.com/document/65de35b8-78ff-4199-b3ab-ce57ea95301b#video)
- [DJIPSDKVideoStreamCheckTool](https://github.com/dji-sdk/DJIPSDKVideoStreamCheckTool)

dji-dev, 2022-01-11:
Agent comment from cheney.zeng in <a href='https://djisdksupport.zendesk.com/agent/tickets/59031'>Zendesk ticket #59031</a>:<br><br>I used your video and the original sample for testing, the video works well here, it`s recommended that you could use the original sample to have a try.
https://github.com/dji-sdk/Payload-SDK
 [image]​
 [image]​<br><a href='https://djisdksupport.zendesk.com/attachments/token/9ZHLhO01kmZCAmR5yDAT1KdpG' target='zd_attach'>inline592030315.png</a><br><a href='https://djisdksupport.zendesk.com/attachments/token/isDtHxenVk3bTMTD2Uo6ym53f' target='zd_attach'>inline488072011.png</a>
<br><p><em><small>°°°</small></em></p>

OpusK, 2022-01-11:
Hi, @dji-dev 

Thanks for testing and responding.
But, my question is for real time camera streaming, not media file streaming.

However, my approach is using the method of transmitting the frame immediately when encoding (25fps) is complete.
Therefore, depending on the encoding time, there may be a difference in the interval time of the frame transmission.
Can this be a problem?

For reference, I confirmed that the files I uploaded and the transcoded files in your samples were completely the same as the binary.

dji-dev, 2022-01-12:
Agent comment from cheney.zeng in <a href='https://djisdksupport.zendesk.com/agent/tickets/59031'>Zendesk ticket #59031</a>:<br><br>The sample could work with your video, it means the SDK part and your video part are working well.
Now when you use the real time camera streaming, there`s no streaming out, I think it`s your code issue, and it needs to debug it by yourself.

It`s recommended to refer to the documentation and the PSDK sample to debug your code.
https://developer.dji.com/document/414e4790-84ea-4c44-aa08-75ce8b616d97
 [image]​<br><a href='https://djisdksupport.zendesk.com/attachments/token/Hhw97P1snyVnwzjAzyJEofT14' target='zd_attach'>inline1762255683.png</a>
<br><p><em><small>°°°</small></em></p>

OpusK, 2022-01-12:
Hi, @dji-dev 

I was using it as your explanation.
As I said earlier, I was doing real-time video streaming through other hardware encoders.
However, since the transmission interval and encoding contents are somewhat different from the existing HW encoding while using the CUDA Encoder, I asked if these parts can affect streaming. 

Anyway, thank you for confirming that there is no problem with the encoding part.

dji-dev, 2022-01-12:
Agent comment from cheney.zeng in <a href='https://djisdksupport.zendesk.com/agent/tickets/59031'>Zendesk ticket #59031</a>:<br><br>You are welcome, if the provided encoded media archive file is from CUDA Encoder, it could work with the sample, which means there is no problem with it.
<br><p><em><small>°°°</small></em></p>

OpusK, 2022-01-12:
Oh, it was just my fault..
Because of compiler optimization, data length was not updated for AUD... so it not sent AUD data.. :(
Anyway, it is solved!
Thanks for support :)

closed 2022-01-12 by OpusK

dji-dev, 2022-01-12:
Agent comment from cheney.zeng in <a href='https://djisdksupport.zendesk.com/agent/tickets/59031'>Zendesk ticket #59031</a>:<br><br>You are welcome, happy to hear that the issue has been fixed, have a nice day.
<br><p><em><small>°°°</small></em></p>