dji-sdk/Payload-SDK #32: Comunication port cannot be changed unlike OSDK 4.1.0

MH2033, 2022-01-21:
Hello
In our project we're using /dev/tty/dev/acm0 as our UART port to communicate with the drone. The project was built on top of OSDK 4.1.0 and we could specify the port in "userconfig.txt".After trying to run the samples of the new payload SDK, the SDK seems to automatically look for /dev/tty/USB0 and that cannot be changed. Any solutions?

dji-dev, 2022-01-21:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/59558'>Zendesk ticket #59558</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

The UART port can be changed in code file: hal_uart.h
https://github.com/dji-sdk/Payload-SDK/blob/7b5f853e877d0f2443bcab8f57101afe12a9a14b/samples/sample_c/platform/linux/manifold2/hal/hal_uart.h#L48

And there is a link for simple guidance:
https://sdk-forum.dji.net/hc/en-us/articles/4408437830041-PSDK-3-0-version-OSDK-OSDK-5-0-introduction-and-guidance

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team
<br><p><em><small>°°°</small></em></p>

MH2033, 2022-01-21:
Thank you this solved the issue

closed 2022-01-21 by MH2033