dji-sdk/Payload-SDK #33: Flying high

jefferycline1, 2022-01-23:


dji-dev, 2022-01-23:
Agent comment from DJI SDK in <a href='https://djisdksupport.zendesk.com/agent/tickets/59606'>Zendesk ticket #59606</a>:<br><br>尊敬的开发者，感谢您联系DJI 大疆创新
由于github不是我们主要的咨询渠道，您的问题可能跟进不及时。我们建议您通过填写表单（ https://djisdksupport.zendesk.com/hc/zh-cn/requests/new ）向我们反馈问题。或者您也可以在论坛发帖，与其它开发者交流。论坛链接：https://djisdksupport.zendesk.com/hc/zh-cn/community/topics

Dear developer, thank you for contacting DJI.
Since github is not our main consultation channel, your questions may not be followed up in time. We recommend that you fill in the form (https://djisdksupport.zendesk.com/hc/en-us/requests/new) to report problems to us. Or you can post in the forum to communicate with other developers. Forum link: https://djisdksupport.zendesk.com/hc/zh-cn/community/topics
<br><p><em><small>°°°</small></em></p>

jefferycline1, 2022-01-23:
DJI SDK app flying high

dji-dev, 2022-01-24:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/59606'>Zendesk ticket #59606</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

"DJI SDK app flying high"
Sorry, I didn't understand your question. There is a developer website about DJI SDK, you can visit it first. and please describe the SDK you are using and the specific functions, will comfirming it for you.

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team
<br><p><em><small>°°°</small></em></p>

closed 2022-02-21 by jefferycline1