dji-sdk/Payload-SDK #34: MOP server cannot be initialized on osdk port

MH2033, 2022-03-16:
The following error message will be generated when trying to initialize MOP server sample on extension port:

[image]

Connection is established via expansion module and uart plus usb

dji-dev, 2022-03-17:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/61627'>Zendesk ticket #61627</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

Sorry for the inconvenience, this problem is a bug of PSDK 3.0, and we have submited to the relevant department for fixing. If there is an update, we will push it to GitHub.

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team
<br><p><em><small>°°°</small></em></p>

MH2033, 2022-03-17:
Thank you for your answer. Can you give us an estimation on when this bug will be fixed? I'm asking since our product relies heavily on this feature.

dji-dev, 2022-03-17:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/61627'>Zendesk ticket #61627</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

Sorry, this may not be able to tell you the specific time at present, we will update and submit this issue again, and we will ask the relevant departments to fix it as soon as possible.

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team
<br><p><em><small>°°°</small></em></p>

MH2033, 2022-03-18:
Thank you for your answer
Regards

MH2033, 2022-06-06:
Any updates regarding this issue?

samanazadpour, 2022-06-15:
@dji-dev we are waiting for you guys. would you please give us an update? Our application is using MOP server and at the moment it is not working! we need to know what is the road map and when it will be fixed.

MH2033, 2022-08-10:
This issue seems to be resolved in release 3.2

closed 2022-08-10 by MH2033