dji-sdk/Payload-SDK #35: File list function halts

MH2033, 2022-03-18:
After running the provided C samples and trying to run the file manager sample, the function responsible for getting the file list halts and never returns. Is it a known issue?
The onboard computer is connected to the drone via extension port

dji-dev, 2022-03-21:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/61725'>Zendesk ticket #61725</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

Please check the link：
https://github.com/dji-sdk/Onboard-SDK/issues/847

If it is the sample issue, you can comment out the file list data lenth checking first.

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team
<br><p><em><small>°°°</small></em></p>

MH2033, 2022-03-21:
Thank you for your answer. 
I've had the similar issue on OSDK v4.1.0 and the link that you shared was helpful but in PSDK i don't have access to source files since only the header files are available.

dji-dev, 2022-03-21:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/61725'>Zendesk ticket #61725</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

Sorry, I overlooked that you are using PSDK 3.0, this is a known bug and not fixed yet. We have reported it to the relevant departments to fix it as soon as possible, sorry for the inconvenience caused to you again.

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team
<br><p><em><small>°°°</small></em></p>

MH2033, 2022-03-28:
Hope this issue will be fixed soon
Regards

samanazadpour, 2022-06-16:
@dji-dev do you have any update on this feature bug? we need a fix as soon as possible

MH2033, 2022-08-10:
This issue seems to be resolved in release 3.2

closed 2022-08-10 by MH2033