dji-sdk/Payload-SDK #38: Max Frequency for RTK and GPS location topics

MH2033, 2022-03-28:
Hi
I was reading the dji_fc_subscription.h file and couldn't find any info about the max subscription freq for getting RTK and GPS location. Can you provide a reference?

dji-dev, 2022-03-28:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/62114'>Zendesk ticket #62114</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

The max subscription freq for getting RTK and GPS location is 5HZ.
We will add notes for those topic, you can refer to the relevant introduction in OSDK 4.1 first, the telemetry subscription frequency is the same
https://github.com/dji-sdk/Onboard-SDK/blob/master/osdk-core/api/inc/dji_telemetry_doc.hpp

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team
<br><p><em><small>°°°</small></em></p>

MH2033, 2022-03-28:
Thank you for your response. 
I saw the link you mentioned before and there's contradiction in the docs. As you can see in the following link it is stated that max subscription freq for GPS and RTK location is 50Hz :  https://developer.dji.com/onboard-sdk/documentation/guides/component-guide-telemetry.html
[image]

dji-dev, 2022-03-28:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/62114'>Zendesk ticket #62114</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

The webpage has updated, and please refer to the code file or this link for specific TOPIC:
https://developer.dji.com/onboard-api-reference/group__telem.html

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team
<br><p><em><small>°°°</small></em></p>

MH2033, 2022-03-29:
Thank you for sharing this link.
Regards

closed 2022-03-29 by MH2033