dji-sdk/Payload-SDK #46: ESP32 lib errors “dangerous relocation: call8: call target out of range”

daz, 2022-04-29:
I’m getting these errors compiling against the ESP32 `libpayloadsdk.a`

```
lib/psdk/libpayloadsdk.a(dji_sdk_state_push.c.o): in function `DjiSdkStatePush_Work':
dji_sdk_state_push.c:(.text+0xb5): dangerous relocation: call8: call target out of range: memset
dji_sdk_state_push.c:(.text+0x121): dangerous relocation: call8: call target out of range: memcpy
dji_sdk_state_push.c:(.text+0x1a1): dangerous relocation: call8: call target out of range: strncpy
lib/psdk/libpayloadsdk.a(dji_high_speed_data_channel.c.o): in function `DjiHighSpeedDataChannel_Init':
dji_high_speed_data_channel.c:(.text+0x11e): dangerous relocation: call8: call target out of range: memcpy
dji_high_speed_data_channel.c:(.text+0x172): dangerous relocation: call8: call target out of range: memcpy
```

The fix is to compile the library with the option "-mlongcalls"

See: https://www.esp32.com/viewtopic.php?t=549

Is it possible for the team to recompile the ESP32 library with this flag?

Thanks!

cmillauriaux, 2022-05-16:
Please, can you fix this quickly ? It's mandatory to run PSDK on ESP32.

Thanks !

daz, 2022-05-19:
Fixed in aaf1a07c437d82960871334e6136a0218d228c8f

closed 2022-05-19 by daz