dji-sdk/Payload-SDK #55: MOP server cannot be initialized on osdk port

samanazadpour, 2022-06-09:
MOP server cannot be initialized on osdk port

dji-dev, 2022-06-16:
Agent comment from Hummels Lei in <a href='https://djisdksupport.zendesk.com/agent/tickets/65735'>Zendesk ticket #65735</a>:<br><br>Sorry，It is a bug of PSDK, We have submitted it and will be fixed.
<br><p><em><small>°°°</small></em></p>

samanazadpour, 2022-06-16:
@dji-dev our project is pending this feature and the file management feature. would you please give us a time frame?
we need this to be fixed as soon as possible since our customers are waiting for our new release.
we cannot release anything until MOP server is fully functional! Please support the developers and make sure your PSDK is ready for onboard computer as soon as possible.

dji-dev, 2022-06-17:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/65735'>Zendesk ticket #65735</a>:<br><br>Sorry for the inconvenience caused to you, because the links between OSDK and PSDK on the M300 are different, and there is indeed a bug when PSDK integrates MOP on OSDK port. We have urged the relevant engineers to fix it as soon as possible.
<br><p><em><small>°°°</small></em></p>

dji-dev, 2022-08-08:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/65735'>Zendesk ticket #65735</a>:<br><br>This issue has been fixed in version 3.2.
<br><p><em><small>°°°</small></em></p>

samanazadpour, 2022-08-10:
This issue is fixed in release 3.2

closed 2022-08-10 by samanazadpour