dji-sdk/Payload-SDK #85: Run camera manager sample FAILED

Needrom, 2023-02-13:
运行PSDK，并选择“l”，开启camera sample的时候出现'''Init usb bulk channel failed, ret:000000EC. '''

具体输出:
'''
[0.008][core]-[Info]-[DjiCore_Init:95) Payload SDK Version : V3.3.0-beta.0-build.1659 
[0.042][adapter]-[Info]-[DjiAccessAdapter_Init:171) Identify aircraft series is Matrice 30 Series 
[0.042][adapter]-[Info]-[DjiAccessAdapter_Init:189) Identify mount position type is Extension Port Type 
[2.056][adapter]-[Info]-[DjiPayloadNegotiate_Start:185) Waiting payload negotiate finish. 
[5.070][adapter]-[Info]-[DjiPayloadNegotiate_Start:185) Waiting payload negotiate finish. 
[6.070][core]-[Info]-[DjiIdentityVerify_UpdatePolicy:411) Updating dji sdk policy file... 
[7.070][core]-[Info]-[DjiIdentityVerify_UpdatePolicy:414) Update dji sdk policy file successfully 
[7.085][core]-[Info]-[DjiCore_Init:163) Identify AircraftType = Matrice 30, MountPosition = Extension Port, SdkAdapterType = None 
[7.086][core]-[Info]-[DjiCore_ApplicationStart:223) Start dji sdk application 
[7.086][user]-[Info]-[DjiUser_ApplicationStart:239) Application start. 

| Available commands:                                                                              |
| [0] Fc subscribe sample - subscribe quaternion and gps data                                      |
| [1] Flight controller sample - take off landing                                                  |
| [2] Flight controller sample - take off position ctrl landing                                    |
| [3] Flight controller sample - take off go home force landing                                    |
| [4] Flight controller sample - take off velocity ctrl landing                                    |
| [5] Flight controller sample - arrest flying                                                     |
| [6] Flight controller sample - set get parameters                                                |
| [7] Hms info sample - get health manger system info                                              |
| [8] Waypoint 2.0 sample - run airline mission by settings (only support on M300 RTK)             |
| [9] Waypoint 3.0 sample - run airline mission by kmz file (not support on M300 RTK)              |
| [a] Gimbal manager sample - rotate gimbal on free mode                                           |
| [b] Gimbal manager sample - rotate gimbal on yaw follow mode                                     |
| [c] Camera stream view sample - display the camera video stream                                  |
| [d] Stereo vision view sample - display the stereo image                                         |
| [e] Start camera all features sample - you can operate the camera on DJI Pilot                   |
| [f] Start gimbal all features sample - you can operate the gimbal on DJI Pilot                   |
| [g] Start widget all features sample - you can operate the widget on DJI Pilot                   |
| [h] Start widget speaker sample - you can operate the speaker on DJI Pilot2                      |
| [i] Start power management sample - you will see notification when aircraft power off            |
| [j] Start data transmission sample - you can send or recv custom data on MSDK demo               |
| [l] Run camera manager sample - shoot photo by the selected camera mounted position              |
| [m] Run camera manager download sample - download camera media file (only support on M3E/M3T)    |

l
[6210.456][user]-[Info]-[DjiTest_CameraManagerRunSample:799) Camera manager sample start 
[6210.456][user]-[Info]-[DjiTest_CameraManagerRunSample:802) --> Step 1: Init camera manager module 
[6210.456][channel]-[Error]-[DjiStreamChannelUsbBulk_AddPort:404) Init usb bulk channel failed, ret:000000EC. 
[6210.456][channel]-[Error]-[DjiStreamChannelUsbBulk_Create:131) Add usb bulk port failed, ret:000000EC. 
[6210.456][downloader]-[Error]-[DjiCameraManager_MediaDownloaderInit:245) Init usb bulk stream channel failed, ret:236. 
[6210.456][cmu]-[Error]-[DjiCameraManager_Init:212) Init media downloader failed, error code: 0x000000E3. 
[6210.456][user]-[Error]-[DjiTest_CameraManagerRunSample:806) Init camera manager failed, error code: 0x000000E3
 
Segmentation fault
'''

有没有哪里可以提示需要排查的。

maurerle, 2025-02-26:
I have the same error:

Don't know how to fix this `Init usb bulk channel failed, ret:000000EC`

maurerle, 2025-03-21:
I asked the DJI support, which responded that one should use `sudo ./bin/dji_sdk_demo_linux_cxx
` or root to solve the `Init usb bulk channel failed` error.
This worked fine. So I hope this helps others as well!