dji-sdk/Payload-SDK #109: 无法接收速度话题消息

Needrom, 2023-07-03:
初始化
```
 djiStat = DjiFcSubscription_SubscribeTopic(DJI_FC_SUBSCRIPTION_TOPIC_VELOCITY, DJI_DATA_SUBSCRIPTION_TOPIC_1_HZ,
DjiTest_FcSubscriptionReceiveVelocityCallback);
//NULL);
    if(djiStat != DJI_ERROR_SYSTEM_MODULE_CODE_SUCCESS) {
        USER_LOG_ERROR("Subscribe topic velocity error");
        cout << "Subscribe topic velocity error" << endl;
        return ;
    } else {
        USER_LOG_ERROR("Subscribe topic velocity success.");
        cout << "Subscribe topic velocity success." << endl;
    }
```

回调:

```

T_DjiReturnCode DjiTest_FcSubscriptionReceiveVelocityCallback(const uint8_t *data, uint16_t dataSize, const T_DjiDataTimestamp *timestamp) {
    T_DjiFcSubscriptionVelocity *velocity = (T_DjiFcSubscriptionVelocity *) data;

    std::cout << "vgx :" << velocity->data.x << " vgy:" << velocity->data.y << " vgz: " << velocity->data.z << std::endl;

    return DJI_ERROR_SYSTEM_MODULE_CODE_SUCCESS;
}

```

dji-dev, 2023-07-03:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/86101'>Zendesk ticket #86101</a>:<br><br>1、检查一下PSDK日志中，订阅接口是不是有报错。
2、不使用回调方式，参考sample中DjiFcSubscription_GetLatestValueOfTopic获取速度值看是否正确？
3、如果是M3行业机型，获取速度目前有bug，待修复。
<br><p><em><small>°°°</small></em></p>

Needrom, 2023-07-03:
> 

对 是M3T， 速度值完全没有，就算是用你们的例程都是没有数值，不显示失败但是没有数值，

dji-dev, 2023-07-03:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/86101'>Zendesk ticket #86101</a>:<br><br>这个要等M3行业版发固件版本了。
<br><p><em><small>°°°</small></em></p>