dji-sdk/Payload-SDK #116: Support for streaming video to controller with E-port expansion board?

uzgit, 2023-08-14:
Will the E-port expansion support streaming video from a connected computer to the controller? (e.g., as in the camera_emu example) Is there a way to get it to work already? I am working with a Jetson Nano, Matrice 350, and E-port expansion board.

dji-dev, 2023-08-15:
Agent comment from shiyao.liao in <a href='https://djisdksupport.zendesk.com/agent/tickets/89039'>Zendesk ticket #89039</a>:<br><br>Dear Developer,

Hello, thank you for contacting DJI.

Thank you for your patience.
In the current PSDK version, the camera_emu example is a function of the PSDK port. If you use the OSDK port, you cannot directly push the stream from the onboard computer to the remote control.

However, in the subsequent PSDK update, the M350 model will support this function. Please pay attention to the official release announcement.

Thank you for your support of DJI products.

Best Regards,
DJI Dajiang innovation SDK technical support
<br><p><em><small>°°°</small></em></p>

uzgit, 2023-09-11:
Hi, when will this be implemented?

bonchan, 2024-06-19:
I add myself to this question @dji-dev