dji-sdk/Payload-SDK #122: Too many build warnings

whyscience, 2023-09-25:
## one bug:
```cpp
case E_DJI_TEST_CAMERA_MANAGER_SAMPLE_SELECT_RESET_CAMERA_SETTINGS: {
            returnCode = DjiCameraManager_ResetCameraSetting(mountPosition);
            if (returnCode != DJI_ERROR_SYSTEM_MODULE_CODE_SUCCESS) {
                USER_LOG_ERROR("reset camera at position %d failed", mountPosition);
                goto exitCameraModule;
            }

            USER_LOG_INFO("Reset camera settings success!");

            break;
        }
```

but in the definition:
```cpp
/**
 * @brief Reset camera settings.
 * @param position: camera mounted position
 * @return Execution result.
 */
T_DjiReturnCode DjiCameraManager_ResetCameraSettings(E_DjiMountPosition position);
```

## also there are tons of build warnings

```
====================[ Build | dji_sdk_demo_linux_cxx | Debug ]==================
/home/eric/.local/share/JetBrains/Toolbox/apps/clion/bin/cmake/linux/x64/bin/cmake --build /home/eric/ct/Payload-SDK-github/cmake-build-debug --target dji_sdk_demo_linux_cxx -j 10
[5/58] Building CXX object samples/sample_c++/platform/linux/manifold2/CMakeFiles/dji_sdk_demo_linux_cxx.dir/__/__/__/module_sample/liveview/dji_camera_stream_decoder.cpp.o
../samples/sample_c++/module_sample/liveview/dji_camera_stream_decoder.cpp: In member function ‘void DJICameraStreamDecoder::decodeBuffer(const uint8_t*, int)’:
../samples/sample_c++/module_sample/liveview/dji_camera_stream_decoder.cpp:210:74: warning: ‘int avcodec_decode_video2(AVCodecContext*, AVFrame*, int*, const AVPacket*)’ is deprecated [-Wdeprecated-declarations]
             avcodec_decode_video2(pCodecCtx, pFrameYUV, &gotPicture, &pkt);
                                                                          ^
In file included from ../samples/sample_c++/module_sample/liveview/dji_camera_stream_decoder.hpp:33:0,
                 from ../samples/sample_c++/module_sample/liveview/dji_camera_stream_decoder.cpp:26:
/usr/include/x86_64-linux-gnu/libavcodec/avcodec.h:4993:5: note: declared here
 int avcodec_decode_video2(AVCodecContext *avctx, AVFrame *picture,
     ^~~~~~~~~~~~~~~~~~~~~
../samples/sample_c++/module_sample/liveview/dji_camera_stream_decoder.cpp:227:72: warning: ‘int avpicture_get_size(AVPixelFormat, int, int)’ is deprecated [-Wdeprecated-declarations]
                     bufSize = avpicture_get_size(AV_PIX_FMT_RGB24, w, h);
                                                                        ^
In file included from ../samples/sample_c++/module_sample/liveview/dji_camera_stream_decoder.hpp:33:0,
                 from ../samples/sample_c++/module_sample/liveview/dji_camera_stream_decoder.cpp:26:
/usr/include/x86_64-linux-gnu/libavcodec/avcodec.h:5653:5: note: declared here
 int avpicture_get_size(enum AVPixelFormat pix_fmt, int width, int height);
     ^~~~~~~~~~~~~~~~~~
../samples/sample_c++/module_sample/liveview/dji_camera_stream_decoder.cpp:229:91: warning: ‘int avpicture_fill(AVPicture*, const uint8_t*, AVPixelFormat, int, int)’ is deprecated [-Wdeprecated-declarations]
                     avpicture_fill((AVPicture *) pFrameRGB, rgbBuf, AV_PIX_FMT_RGB24, w, h);
                                                                                           ^
In file included from ../samples/sample_c++/module_sample/liveview/dji_camera_stream_decoder.hpp:33:0,
                 from ../samples/sample_c++/module_sample/liveview/dji_camera_stream_decoder.cpp:26:
/usr/include/x86_64-linux-gnu/libavcodec/avcodec.h:5638:5: note: declared here
 int avpicture_fill(AVPicture *picture, const uint8_t *ptr,
     ^~~~~~~~~~~~~~
../samples/sample_c++/module_sample/liveview/dji_camera_stream_decoder.cpp:246:24: warning: ‘void av_free_packet(AVPacket*)’ is deprecated [-Wdeprecated-declarations]
     av_free_packet(&pkt);
                        ^
In file included from ../samples/sample_c++/module_sample/liveview/dji_camera_stream_decoder.hpp:33:0,
                 from ../samples/sample_c++/module_sample/liveview/dji_camera_stream_decoder.cpp:26:
/usr/include/x86_64-linux-gnu/libavcodec/avcodec.h:4650:6: note: declared here
 void av_free_packet(AVPacket *pkt);
      ^~~~~~~~~~~~~~
[9/58] Building C object samples/sample_c++/platform/linux/manifold2/CMakeFiles/dji_sdk_demo_linux_cxx.dir/__/__/__/__/sample_c/module_sample/camera_emu/dji_media_file_manage/dji_media_file_core.c.o
../samples/sample_c/module_sample/camera_emu/dji_media_file_manage/dji_media_file_core.c:49:9: warning: initialization from incompatible pointer type [-Wincompatible-pointer-types]
         DjiMediaFile_GetDataOrigin_JPG,
         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
../samples/sample_c/module_sample/camera_emu/dji_media_file_manage/dji_media_file_core.c:49:9: note: (near initialization for ‘s_mediaFileOpt[0].getDataOrgFunc’)
../samples/sample_c/module_sample/camera_emu/dji_media_file_manage/dji_media_file_core.c:65:9: warning: initialization from incompatible pointer type [-Wincompatible-pointer-types]
[... 87 lines cut ...]
[58/58] Linking CXX executable bin/dji_sdk_demo_linux_cxx
-- Cannot Find OPUS
-- Found LIBUSB installed in the system
--  - Includes: /usr/include/libusb-1.0
--  - Libraries: /usr/lib/x86_64-linux-gnu/libusb-1.0.so

dji_sdk_demo_linux_cxx...
-- Found OpenCV installed in the system, will use it to display image in AdvancedSensing APIs
--  - Includes: /usr/local/include;/usr/local/include/opencv
--  - Libraries: opencv_calib3d;opencv_core;opencv_dnn;opencv_features2d;opencv_flann;opencv_highgui;opencv_imgcodecs;opencv_imgproc;opencv_ml;opencv_objdetect;opencv_photo;opencv_shape;opencv_stitching;opencv_superres;opencv_video;opencv_videoio;opencv_videostab;opencv_viz;opencv_aruco;opencv_bgsegm;opencv_bioinspired;opencv_ccalib;opencv_cvv;opencv_datasets;opencv_dnn_objdetect;opencv_dpm;opencv_face;opencv_freetype;opencv_fuzzy;opencv_hdf;opencv_hfs;opencv_img_hash;opencv_line_descriptor;opencv_optflow;opencv_phase_unwrapping;opencv_plot;opencv_reg;opencv_rgbd;opencv_saliency;opencv_sfm;opencv_stereo;opencv_structured_light;opencv_surface_matching;opencv_text;opencv_tracking;opencv_xfeatures2d;opencv_ximgproc;opencv_xobjdetect;opencv_xphoto
-- Found FFMPEG installed in the system
--  - Includes: /usr/include/x86_64-linux-gnu
--  - Libraries: /usr/lib/x86_64-linux-gnu/libavcodec.so;/usr/lib/x86_64-linux-gnu/libavformat.so;/usr/lib/x86_64-linux-gnu/libavutil.so;/usr/lib/x86_64-linux-gnu/libswresample.so;/usr/lib/x86_64-linux-gnu/libswscale.so;/usr/lib/x86_64-linux-gnu/libtheora.so;/usr/lib/x86_64-linux-gnu/libz.so
--  - Version:  3.4.11-0ubuntu0.1 
-- Cannot Find OPUS
-- Found LIBUSB installed in the system
--  - Includes: /usr/include/libusb-1.0
--  - Libraries: /usr/lib/x86_64-linux-gnu/libusb-1.0.so
-- Configuring done
-- Generating done
-- Build files have been written to: /home/eric/ct/Payload-SDK-github/cmake-build-debug

Build finished

```

whyscience, 2023-09-25:
also this 
```
hal_usb_bulk.c:106:13: note: ‘snprintf’ output between 5 and 68 bytes into a destination of size 64
             snprintf(usbBulk2EpOutFd, sizeof(usbBulk2EpOutFd), "%s/ep1",
             ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
                      linkConfig.usbBulkConfig.usbBulk2DeviceName);
```

dji-dev, 2023-09-27:
Agent comment from shiyao.liao in <a href='https://djisdksupport.zendesk.com/agent/tickets/91743'>Zendesk ticket #91743</a>:<br><br>Dear Developer,

Hello, thank you for contacting DJI.

Sorry for the inconvenience. Which development platform and tool chain are you using for compilation? I will feed this information back to the R&D team.

Thank you for your support of DJI products.

Best Regards,
DJI Dajiang innovation SDK technical support
<br><p><em><small>°°°</small></em></p>