dji-sdk/Payload-SDK #125: 在第三方arm64主机编译失败 undefined reference to `DjiCameraManager_ResetCameraSettings'

SJJ-dot, 2023-10-16:
报错信息如下，注释掉报错这一行代码后，`test_camera_manager.c` 文件编译成功
```
CMakeFiles/dji_sdk_demo_linux.dir/__/__/__/module_sample/camera_manager/test_camera_manager.c.o: In function `DjiTest_CameraManagerRunSample':
test_camera_manager.c:(.text+0x3f18): undefined reference to `DjiCameraManager_ResetCameraSettings'
collect2: error: ld returned 1 exit status
samples/sample_c/platform/linux/manifold2/CMakeFiles/dji_sdk_demo_linux.dir/build.make:1266: recipe for target 'bin/dji_sdk_demo_linux' failed
make[2]: *** [bin/dji_sdk_demo_linux] Error 1
CMakeFiles/Makefile2:122: recipe for target 'samples/sample_c/platform/linux/manifold2/CMakeFiles/dji_sdk_demo_linux.dir/all' failed
make[1]: *** [samples/sample_c/platform/linux/manifold2/CMakeFiles/dji_sdk_demo_linux.dir/all] Error 2
Makefile:129: recipe for target 'all' failed
make: *** [all] Error 2
```

dji-dev, 2023-10-17:
Agent comment from shiyao.liao in <a href='https://djisdksupport.zendesk.com/agent/tickets/92868'>Zendesk ticket #92868</a>:<br><br>尊敬的开发者，

您好，感谢您联系DJI 大疆创新。

很抱歉给您带来不好的体验，请问一下您使用的是什么平台，方便把平台信息同步给我们吗？

感谢您对大疆产品的支持！祝您一切顺利！

Best Regards,
DJI 大疆创新SDK技术支持
<br><p><em><small>°°°</small></em></p>

SJJ-dot, 2023-10-17:
> Agent comment from shiyao.liao in [Zendesk ticket #92868](https://djisdksupport.zendesk.com/agent/tickets/92868):尊敬的开发者，
> 
> 您好，感谢您联系DJI 大疆创新。
> 
> 很抱歉给您带来不好的体验，请问一下您使用的是什么平台，方便把平台信息同步给我们吗？
> 
> 感谢您对大疆产品的支持！祝您一切顺利！
> 
> Best Regards, DJI 大疆创新SDK技术支持
> 
> _°°°_

arm64 主机，无GPU，在docker Ubuntu 18.04中编译，