dji-sdk/Payload-SDK #127: 使用bootloader跳转，在APP中为何没有重新配置中断向量偏移？

475241512, 2023-10-23:
是需要自己配置还是不需要？源码不改，直接编译下载应该会跑飞吧？

475241512, 2023-10-24:
freertos处理

closed 2023-10-24 by 475241512

dji-dev, 2023-10-24:
Agent comment from xiaorong.zheng in <a href='https://djisdksupport.zendesk.com/agent/tickets/93303'>Zendesk ticket #93303</a>:<br><br>freertos需要手动进行配置。
<br><p><em><small>°°°</small></em></p>

475241512, 2023-10-24:
谢谢。运行没问题，但是不知道是在freertos中哪里配置中断向量偏移的，以前只是简单粗暴的在初始化前重新映射  - .-
`SCB->VTOR = FLASH_BASE | APPLICATION_ADDRESS;`
MDK - Target - IROM1 - Start  0x8010000
@dji-dev 能否请教一下在什么位置配置，谢谢 :)

dji-dev, 2023-10-24:
Agent comment from xiaorong.zheng in <a href='https://djisdksupport.zendesk.com/agent/tickets/93303'>Zendesk ticket #93303</a>:<br><br>这个要根据您自己的芯片手册进行配置，网上其实有很多可供参考的资料，这个可能需要您自己学习一下。
<br><p><em><small>°°°</small></em></p>

475241512, 2023-10-25:
谢谢！看到了在启动文件中配置了 ：）