dji-sdk/Payload-SDK #133: Dji Dock and Payload SDK

abeldixon, 2023-11-10:
I would like to understand more about the behaviour of the Payload SDK when it used along with the DJI Dock, I have tried giving go to location to DJI Dock and then mission from the Payload SDK, but the mission pause is not working from PSDK, I am running the setup in NVIDIA Jetson.

dji-dev, 2023-11-10:
Agent comment from shiyao.liao in <a href='https://djisdksupport.zendesk.com/agent/tickets/94729'>Zendesk ticket #94729</a>:<br><br>Dear Developer,

Hello, thank you for contacting DJI.

I'm sorry to bring you a bad experience. I would like to confirm with you that what you mentioned does not work. Is there any error reported in PSDK?

Thank you for your support of DJI products.

Best Regards,
DJI Dajiang innovation SDK technical support
<br><p><em><small>°°°</small></em></p>

abeldixon, 2023-11-15:
[image]

I am getting the following error is there any possible resolution for this ?, it seems the psdk is unable to get the current state of the drone.
When I gave mission from PSDK I am able to pause from the Cloud API but not able to resume. (319044 : Task completed,Unable to resume), and Pausing/Resuming/Stopping from the PSDK is responding with the cmd retry failed when the mission is given from the hover state after take off from the dock.

dji-dev, 2023-11-16:
Agent comment from shiyao.liao in <a href='https://djisdksupport.zendesk.com/agent/tickets/94729'>Zendesk ticket #94729</a>:<br><br>Dear Developer,

Hello, thank you for contacting DJI.

I saw that the error code…000E1 was reported in the error log. This means that the operation has timed out, which is usually caused by the link. At this time, you can check whether your Uart and BULK links are normal.

Thank you for your support of DJI products.

Best Regards,
DJI Dajiang innovation SDK technical support
<br><p><em><small>°°°</small></em></p>

abeldixon, 2023-11-16:
Yes, the UART connection is still active because it's launching the sample application and establishing the connection with drone with user option selection ?
The mission execution and manual controls are working fine but the pause/resume is not working.

dji-dev, 2023-11-17:
Agent comment from shiyao.liao in <a href='https://djisdksupport.zendesk.com/agent/tickets/94729'>Zendesk ticket #94729</a>:<br><br>Dear Developer,

Hello, thank you for contacting DJI.

Could you please send us the complete PSDK operation log? Let's take a look at the communication process in PSDK.

Thank you for your support of DJI products.

Best Regards,
DJI Dajiang innovation SDK technical support
<br><p><em><small>°°°</small></em></p>