dji-sdk/Payload-SDK #141: PSDK3.7 负载获取摄像机 H264 视频流问题

lxue1986, 2024-01-28:
机场：大疆机场
无人机：M30T（机场版）

问题：
        经过测试多条航线，负载启动之后获取不到摄像机的H264视频流（通过DjiLiveview_StartH264Stream注册的回调函数一直没有获取到视频流数据），但是我们发现一个奇怪的现象，当无人机飞完最后一个航点返航的时候，负载却可以正常获取到摄像机的H264视频流数据，多条不同的航线都有相同的规律，请问这种情况会是什么原因造成的？

备注：在负载获取不到摄像机H264视频流的时间段内，系统后台无人机视角的视频直播画面是正常的

测试结果：
[image]

dji-dev, 2024-01-29:
Agent comment from xiaorong.zheng in <a href='https://djisdksupport.zendesk.com/agent/tickets/100045'>Zendesk ticket #100045</a>:<br><br>尊敬的开发者，
您好，感谢您联系DJI 大疆创新。

麻烦您修改代码测试，在第一次获取流失败后，再次调用DjiLiveview_StartH264Stream拉流进行测试。

希望我们的解决方案能够帮到您，感谢您的邮件，祝您生活愉快！

Best Regards,
DJI 大疆创新SDK技术支持
<br><p><em><small>°°°</small></em></p>

lxue1986, 2024-01-29:
> Agent comment from xiaorong.zheng in [Zendesk ticket #100045](https://djisdksupport.zendesk.com/agent/tickets/100045):尊敬的开发者， 您好，感谢您联系DJI 大疆创新。
> 
> 麻烦您修改代码测试，在第一次获取流失败后，再次调用DjiLiveview_StartH264Stream拉流进行测试。
> 
> 希望我们的解决方案能够帮到您，感谢您的邮件，祝您生活愉快！
> 
> Best Regards, DJI 大疆创新SDK技术支持
> 
> _°°°_

好的，感谢您的回复，我们测试一下

Jessicahjj, 2024-01-30:
按照上面的方法再次测试后，问题没有得到解决。

jingbo18, 2024-01-31:
> Contributor
请问PSDK有没有类似的销毁读取相机码流的线程的函数接口？
[image]

dji-dev, 2024-01-31:
Agent comment from xiaorong.zheng in <a href='https://djisdksupport.zendesk.com/agent/tickets/100045'>Zendesk ticket #100045</a>:<br><br>请先使用最新PSDK代码，在第一次获取流失败后，先调用DjiLiveview_StopH264Stream，再调用DjiLiveview_StartH264Stream拉流。如果问题依然存在，就需要您拉日志提单分析。
<br><p><em><small>°°°</small></em></p>