dji-sdk/Payload-SDK #142: Use M30 indoor

stevedanomodolor, 2024-02-05:
We are planning on integrating the M30 for indoor uses but the position data VO_position requires the gps to be available. Although this topic works without gps, i am afraid that if gps data becomes available when flying indoor, this can cause a jump in the localization data which can lead to the drone to crash. 
Are they any available topics that only inlcudes vio topics.

dji-dev, 2024-02-07:
Agent comment from xiaorong.zheng in <a href='https://djisdksupport.zendesk.com/agent/tickets/100598'>Zendesk ticket #100598</a>:<br><br>In indoor environments where GPS is unavailable, it is recommended to solely rely on the Attitude Control Interface. To simulate such conditions, you can encase the GPS module to block its reception of GPS signals.
<br><p><em><small>°°°</small></em></p>

stevedanomodolor, 2024-02-07:
Where is the location of the gps and what can i use to block it

closed 2024-02-09 by stevedanomodolor

dji-dev, 2024-02-12:
Agent comment from xiaorong.zheng in <a href='https://djisdksupport.zendesk.com/agent/tickets/100598'>Zendesk ticket #100598</a>:<br><br>The easiest way to temporarily block GPS signals on your drone is by wrapping its dome antennas in aluminum foil.<br><a href='https://djisdksupport.zendesk.com/attachments/token/Yvws2h3X7Nyi6UkhOdNsmgJFu' target='zd_attach'>image.png</a>
<br><p><em><small>°°°</small></em></p>