dji-sdk/Payload-SDK #144: DjiCore_DeInit segmentation fault

biancabnd, 2024-02-19:
Hello, 
First of all, thank you for maintaining and upgrading this SDK!
While using it, I found myself with the need of shutting down all modules cleanly and thus, using the DjiCore_DeInit() function.
However, the node where I launch this simply dies when I call core deinit. I thought this could be something on my end, so I tried to force the same method on the C Samples you provide. The result I obtain is the following:
[image]

Am I missing something or do I need to do something else before calling DjiCore_DeInit()? How can I avoid having the segmentation fault each time I shut down the psdk application?

Thank you in advance.

dji-dev, 2024-02-20:
Agent comment from Leon in <a href='https://djisdksupport.zendesk.com/agent/tickets/100991'>Zendesk ticket #100991</a>:<br><br>Dear Developer,

Hello, thank you for contacting DJI.

You can try to follow the order in DjiUser_CleanSystemEnvironment() function for de-initialization.

Thank you for your support of DJI products.

Best Regards,
DJI Dajiang innovation SDK technical support
<br><p><em><small>°°°</small></em></p>

biancabnd, 2024-02-20:
Hello. 
Thank you for your answer. The screenshot I have sent you is a result of calling directly the `DjiUser_CleanSystemEnvironment`() function. As you can see, there seems to be a problem with the `Dji_PowerManagement_DeInit`, and thus the `DjiUser_CleanSystemEnvironment`() function fails.

dji-dev, 2024-02-20:
Agent comment from Leon in <a href='https://djisdksupport.zendesk.com/agent/tickets/100991'>Zendesk ticket #100991</a>:<br><br>Dear Developer,

Hello, thank you for contacting DJI.

Which version of PSDK are you using? We may need to further confirm this function.

Thank you for your support of DJI products.

Best Regards,
DJI Dajiang innovation SDK technical support
<br><p><em><small>°°°</small></em></p>

biancabnd, 2024-02-20:
I am using the latest one, v3.8.

dji-dev, 2024-02-21:
Agent comment from Leon in <a href='https://djisdksupport.zendesk.com/agent/tickets/100991'>Zendesk ticket #100991</a>:<br><br>Dear Developer,

Hello, thank you for contacting DJI.

Thank you for the information you provided. We have confirmed that PSDK will start related function modules when it is started. These modules may re-create threads, which will cause some resource lock restrictions when you close the main thread. May I ask what your demand scenario is? Do you want to turn off all module functions?

Thank you for your support of DJI products.

Best Regards,
DJI Dajiang innovation SDK technical support
<br><p><em><small>°°°</small></em></p>

biancabnd, 2024-02-21:
Hi again, 

Yes. What I try to do is:

1. Initialize all modules (`DjiCore_Init`, `DjiFcSubscription_Init`, `DjiCameraManager_Init`, etc). This works perfectly. 
2. Do something with the data, tests, etc - All good here 
3. Shutdown the PSDK application and all modules cleanly. For this I call Deinit functions of the modules, `DjiCameraManager_DeInit`, `DjiFcSubscription_DeInit` which exit with no error. Lastly, I want to call `DjiCore_DeInit`() and this always makes the node die not cleanly, so with segmentation fault. 

Maybe there is something else I need to stop before calling the `DjiCore_DeInit`() function. But I've checked `DjiUser_CleanSystemEnvironment`() from the samples, and this method does not exit cleanly as well. 

My question is, how can I shutdown the entire PSDK application cleanly and turning off all modules?

dji-dev, 2024-02-22:
Agent comment from Leon in <a href='https://djisdksupport.zendesk.com/agent/tickets/100991'>Zendesk ticket #100991</a>:<br><br>Dear Developer,

Hello, thank you for contacting DJI.

Regarding your request, we may have to confirm with the PSDK R&D team to see if the interface is working properly. We will contact you again soon. Due to our large consultation volume, there may be omissions. If it takes three working days It has not been updated for you, please contact me again. Thank you for your understanding~

Thank you for your support of DJI products.

Best Regards,
DJI Dajiang innovation SDK technical support
<br><p><em><small>°°°</small></em></p>

biancabnd, 2024-03-08:
Hey, do you have an update on this issue?

Thank you.

dji-dev, 2024-03-11:
Agent comment from Leon in <a href='https://djisdksupport.zendesk.com/agent/tickets/100991'>Zendesk ticket #100991</a>:<br><br>Dear Developer,

Hello, thank you for contacting DJI.

Thank you for your patience. We have confirmed with the R&D team that there is a bug in this interface, and it is currently being scheduled for repair. We have not yet received an accurate reply on the specific date of repair. We are simultaneously confirming for you whether there is a temporary method to solve it. Regarding this issue, we will contact you again soon.

Thank you for your support of DJI products.

Best Regards,
DJI Dajiang innovation SDK technical support
<br><p><em><small>°°°</small></em></p>

biancabnd, 2024-04-08:
Hello!
Any updates on this? Are you planning a new release with some of these bug fixes?

Thank you.