dji-sdk/Payload-SDK #147: UART0 connection failed

hgdly, 2024-02-29:
Hi,

I've run the sample code provided and on execution I get this error:
```
[0.005][core]-[Info]-[DjiCore_Init:101) Payload SDK Version : V3.8.0-beta.0-build.1929 
[2.198][adapter]-[Info]-[DjiAccessAdapter_Init:212) Try identify UART0 connection failed. Probably because SDK adapter or aircraft not finish init or UART connect error. 
[4.426][adapter]-[Info]-[DjiAccessAdapter_Init:212) Try identify UART0 connection failed. Probably because SDK adapter or aircraft not finish init or UART connect error.
```
I don't understand what the LINUX_UART_DEV1 and LINUX_UART_DEV2 configurations mean. Could you explain them to me and tell me how to find them? 

Thanks in advance, 
Hugo

dji-dev, 2024-03-01:
Agent comment from Leon in <a href='https://djisdksupport.zendesk.com/agent/tickets/101744'>Zendesk ticket #101744</a>:<br><br>Dear Developer,

Hello, thank you for contacting DJI.

May I ask which model are you using? If it is an M300 model, you need to configure DEV1 and DEV2. For other models, you do not need to pay attention to DEV2.
The reason why M300 needs to be configured with DEV2 is because the serial port channel of M300 not only uses Uart, but also uses some channels in the USB link as the serial port function, so it needs to be configured with two serial ports.

Thank you for your support of DJI products.

Best Regards,
DJI Dajiang innovation SDK technical support
<br><p><em><small>°°°</small></em></p>

hgdly, 2024-03-01:
> Agent comment from Leon in [Zendesk ticket #101744](https://djisdksupport.zendesk.com/agent/tickets/101744):Dear Developer,
> 
> Hello, thank you for contacting DJI.
> 
> May I ask which model are you using? If it is an M300 model, you need to configure DEV1 and DEV2. For other models, you do not need to pay attention to DEV2. The reason why M300 needs to be configured with DEV2 is because the serial port channel of M300 not only uses Uart, but also uses some channels in the USB link as the serial port function, so it needs to be configured with two serial ports.
> 
> Thank you for your support of DJI products.
> 
> Best Regards, DJI Dajiang innovation SDK technical support
> 
> _°°°_

I'm using M350 model.

hgdly, 2024-03-25:
I still haven't managed to solve my problem, so I'm sending another message. Thanks for your reply !

koikonomou, 2024-03-28:
I have the same problem with M350 model ! Did you manage to solve it ?

hgdly, 2024-03-29:
> I have the same problem with M350 model ! Did you manage to solve it ?

No, not yet

koikonomou, 2024-03-29:
Found a solution to the serial connection problem. 
The issue was with the Dupont lines. 
This resolved the issue for me. 

1.     Replace old Dupont lines with new ones.
2.     Ensure connections are correct: Developer Kit TX to Serial Module RX, Developer Kit RX to Serial Module TX, and both GNDs connected.
Hope it helps!

hgdly, 2024-04-02:
Hi, I don't understand exactly what you've done. How did you find out what to put in the LINUX_UART_DEV1 value?

Lmixq, 2024-05-24:
I have met the same problem recently. But I have solved this by a chance.
`I use the E-Port Development Kit instead of SDK Round Ribbon Cable to connet with M350.`
And notice the AB side of the port.
[https://dl.djicdn.com/downloads/matrice_350_rtk/Matrice_350_RTK_E_Port_Development_Kit_Product_Information.pdf](url)