dji-sdk/Payload-SDK #148: zHealth flag of T_DjiFcSubscriptionPositionVO is always 0

biancabnd, 2024-02-29:
Hello, 

When subscribing to the `T_DjiFcSubscriptionPositionVO` data using the latest version of Payload-SDK (v3.8) I see that the `zHealth` flag is always `zHealth = 0 `when using an DJI M300 RTK drone in the field. However, the x and y health flag report a 1. See image below:

[image]

The `zHealth` flag is only 1 when connected to the DJI Assistant 2 simulator. 
Is there any reason why this flag is always 0 when used in a real set-up?

Thank you.

dji-dev, 2024-03-01:
Agent comment from Leon in <a href='https://djisdksupport.zendesk.com/agent/tickets/101766'>Zendesk ticket #101766</a>:<br><br>Dear Developer,

Hello, thank you for contacting DJI.

Have you installed the load? This *_health indicates that a certain VO sensor is not working. We want to confirm if your load is blocking the downward-looking sensor.

Thank you for your support of DJI products.

Best Regards,
DJI Dajiang innovation SDK technical support
<br><p><em><small>°°°</small></em></p>

biancabnd, 2024-03-01:
Hello, 

There is one load on top of the drone, but underneath there is only the gimbal with a H20 camera attached. Nothing else. 
It should not be blocking the downward-looking sensor. 

Thank you.

dji-dev, 2024-03-04:
Agent comment from Leon in <a href='https://djisdksupport.zendesk.com/agent/tickets/101766'>Zendesk ticket #101766</a>:<br><br>Dear Developer,

Hello, thank you for contacting DJI.

Thank you very much for your feedback. We think this is a bug. We mounted the H20 camera on load position 1 and tried to subscribe. You can see the attached picture. We also see that zHealth is always 0. We will give feedback to the R&D team. Fix this problem as soon as possible. Once there is a repair plan or the repair is completed, we will synchronize this message to you again.

Thank you for your support of DJI products.

Best Regards,
DJI Dajiang innovation SDK technical support<br><a href='https://djisdksupport.zendesk.com/attachments/token/a9mi1I9Avd1cHSKtTLhsqWBE2' target='zd_attach'>image.png</a>
<br><p><em><small>°°°</small></em></p>