dji-sdk/Payload-SDK #151: PSDK commands timesout after multiple calls M30

stevedanomodolor, 2024-03-08:
For bit of context, currently we use flying the m30 with the psdk software. After running multiple missions with streaming, we notice that all of a sudden, randomly the service call to the psdk times out. I also noticed this behavior when I set the camera source to the m30T based on the enum in the payload sdk.  It seems that the enum is incorrect and that the zoom for the m30/m30T is 2 like the h20 while the wide is 1. Please confirm if this is indeed the case. 

```
typedef enum {
    DJI_LIVEVIEW_CAMERA_SOURCE_DEFAULT = 0,
    DJI_LIVEVIEW_CAMERA_SOURCE_H20_WIDE = 1,
    DJI_LIVEVIEW_CAMERA_SOURCE_H20_ZOOM = 2,
    DJI_LIVEVIEW_CAMERA_SOURCE_H20T_WIDE = 1,
    DJI_LIVEVIEW_CAMERA_SOURCE_H20T_ZOOM = 2,
    DJI_LIVEVIEW_CAMERA_SOURCE_H20T_IR = 3,
    DJI_LIVEVIEW_CAMERA_SOURCE_H20N_WIDE = 1,
    DJI_LIVEVIEW_CAMERA_SOURCE_H20N_ZOOM = 2,
    DJI_LIVEVIEW_CAMERA_SOURCE_H20N_IR = 3,
    DJI_LIVEVIEW_CAMERA_SOURCE_M30_ZOOM = 1,
    DJI_LIVEVIEW_CAMERA_SOURCE_M30_WIDE = 2,
    DJI_LIVEVIEW_CAMERA_SOURCE_M30T_ZOOM = 1,
    DJI_LIVEVIEW_CAMERA_SOURCE_M30T_WIDE = 2,
    DJI_LIVEVIEW_CAMERA_SOURCE_M30T_IR = 3,
    DJI_LIVEVIEW_CAMERA_SOURCE_M3E_VIS = 1,
    DJI_LIVEVIEW_CAMERA_SOURCE_M3T_VIS = 1,
    DJI_LIVEVIEW_CAMERA_SOURCE_M3T_IR = 2,
} E_DjiLiveViewCameraSource;
```

I am not sure it is a problem related to the baudrate or the frequency we set for each of the subscriber. Currently for the M30, we use the following information. Also, it seem the documentation on your payload page  on the frequency and topics available for each drone is incorrect ->https://developer.dji.com/doc/payload-sdk-tutorial/en/function-set/basic-function/info-management.html
```

      baudrate: "921600"
      data_frequency:
        imu: 50
        attitude: 50
        acceleration: 50
        velocity: 50
        angular_velocity: 50
        position: 50
        altitude: 50
        gps_fused_position: 10
        gps_data: 10
        rtk_data: 5
        magnetometer: 50
        rc_channels_data: 1
        gimbal_data: 50
        flight_status: 1
        battery_level: 1
        control_information: 1

````

dji-dev, 2024-03-11:
Agent comment from Leon in <a href='https://djisdksupport.zendesk.com/agent/tickets/102367'>Zendesk ticket #102367</a>:<br><br>Dear Developer,

Hello, thank you for contacting DJI.

the m30/m30T is 2 like the h20 while the wide is 1
This is normal and by design.

If your command times out, it is likely that your serial port cannot handle the large amount of data. We noticed that you subscribed to too many TOPICs, and the frequency is generally 50Hz. We recommend that you lower the frequency and try to see if there is an improvement, or reduce the number of TOPICs you subscribe to.

In addition, can you upload your PSDK operation log? File containing Debug information.

Thank you for your support of DJI products.

Best Regards,
DJI Dajiang innovation SDK technical support
<br><p><em><small>°°°</small></em></p>

stevedanomodolor, 2024-03-11:
how can i get the PSDK operation log?

stevedanomodolor, 2024-03-11:
> Hello, thank you for contacting DJI.

stevedanomodolor, 2024-03-12:
I am saying in your code https://github.com/dji-sdk/Payload-SDK/blob/db9d05ee2b3289b957522d49b919861d33020c74/psdk_lib/include/dji_liveview.h#L65 it is wrong, it is put  the other way around.

dji-dev, 2024-03-13:
Agent comment from Leon in <a href='https://djisdksupport.zendesk.com/agent/tickets/102367'>Zendesk ticket #102367</a>:<br><br>Dear Developer,

Hello, thank you for contacting DJI.

Thank you for your patience and additional information. I'm very sorry that I didn't fully understand what you meant before. I fully understand it now. There is a problem with the corresponding numbers in the video source column of the M30's zoom camera and wide-angle camera. I just tested it. The acquisition of these two video sources is corresponding and normal. It should not affect your use currently.
Regarding your feedback, we will simultaneously send it to the R&D team for confirmation and correction. Thank you again~!

Thank you for your support of DJI products.

Best Regards,
DJI Dajiang innovation SDK technical support
<br><p><em><small>°°°</small></em></p>