dji-sdk/Payload-SDK #152: DjiCameraManager SetPointThermometryCoordinat fails

RPS98, 2024-03-21:
# Setup
Aircraft: **DJI Matrice 300 RTK** in v9.00.0503
Payload: **DJI ZENMUSE H20T** v7.00.01.05
PSDK Version: **v3.8**
Connection: **DJI E-Port**

On the controller, I can tap on a point on the thermal image and obtain its temperature.

# Issue
I'm trying to obtain the temperature of a point on the image. To do this, I call the [`DjiCameraManager_SetPointThermometryCoordinate`](https://developer.dji.com/doc/payload-sdk-api-reference/en/module/camera-manager.html#function-djicameramanager-setpointthermometrycoordinate) function to set the coordinates of the point on the image, and then I call [`DjiCameraManager_GetPointThermometryData`](https://developer.dji.com/doc/payload-sdk-api-reference/en/module/camera-manager.html#function-djicameramanager-getpointthermometrydata) to obtain the temperature of that point.

When I call the [`DjiCameraManager_SetPointThermometryCoordinate`](https://developer.dji.com/doc/payload-sdk-api-reference/en/module/camera-manager.html#function-djicameramanager-setpointthermometrycoordinate) function, I receive a **DJI Error Code 21000000E0**. Consequently, when I call the [`DjiCameraManager_GetPointThermometryData`](https://developer.dji.com/doc/payload-sdk-api-reference/en/module/camera-manager.html#function-djicameramanager-getpointthermometrydata) function, I don't receive any information.

dji-dev, 2024-03-22:
Agent comment from Leon in <a href='https://djisdksupport.zendesk.com/agent/tickets/103289'>Zendesk ticket #103289</a>:<br><br>Dear RPS98,

Greetings! We appreciate your contact with DJI - Da-Jiang Innovations.

We apologize for any confusion caused. Upon our confirmation, the H20T camera does not support this interface. Currently, this interface is only supported by the M30T, M3T, and M3TD models. You can refer to the following link for more information:

https://developer.dji.com/doc/payload-sdk-tutorial/en/function-set/basic-function/basic-camera-management.html

Thank you for your support of DJI products! We wish you all the best!

Best Regards,

DJI - Da-Jiang Innovations SDK Technical Support
<br><p><em><small>°°°</small></em></p>

RPS98, 2024-04-05:
Thank you very much for the help!

closed 2024-04-05 by RPS98