dji-sdk/Payload-SDK #154: "I want to receive live streaming video from a DJI drone."

MACROstd, 2024-03-26:
"I am currently using the Mavic Enterprise Series 3T model, and I want to receive live streaming video in an Ubuntu 22.04 environment and perform object detection through a computer vision recognition model, but I don't know which SDK to use."

dji-dev, 2024-03-26:
Agent comment from Leon in <a href='https://djisdksupport.zendesk.com/agent/tickets/103590'>Zendesk ticket #103590</a>:<br><br>Dear MACROstd,

Greetings,

Thank you for reaching out to DJI - Da-Jiang Innovations.

We appreciate your patience. Regarding your request, we recommend that you use the Payload SDK (PSDK) for development. PSDK can assist you in obtaining the video stream information from the drone. You can refer to the following articles for more information:

https://developer.dji.com/doc/payload-sdk-tutorial/en/tutorial-map.html

Thank you for your support of DJI products! We wish you every success!

Best Regards,
DJI - Da-Jiang Innovations SDK Technical Support
<br><p><em><small>°°°</small></em></p>

MACROstd, 2024-03-27:
"Does this mean I can also receive the drone's screen wirelessly?

MACROstd, 2024-03-27:
"I'm currently trying to directly connect a drone to my desktop, but only the ports /dev/sg1,2,3 are recognized. In other references using ros2, it is recognized as usb0, acm0, etc. Why is that?"

dji-dev, 2024-03-27:
Agent comment from Leon in <a href='https://djisdksupport.zendesk.com/agent/tickets/103590'>Zendesk ticket #103590</a>:<br><br>Dear MACROstd,

Hello, thank you for contacting DJI.

》"Does this mean I can also receive the drone's screen wirelessly?
If you simply use PSDK, you need a wired connection to obtain it.

》"I'm currently trying to directly connect a drone to my desktop, but only the ports /dev/sg1,2,3 are recognized. In other references using ros2, it is recognized as usb0, acm0, etc. Why is that ?"
Is it convenient for you to send us the physical diagram of the hardware connection? So that we can better confirm this issue for you, have you used the E-port kit to connect?

Thank you for your support of DJI products! I wish you all the best!

Best Regards,
DJI DJI SDK technical support
<br><p><em><small>°°°</small></em></p>

macrodk, 2024-04-23:
I bought the e port developer kit and connected it, but when I use it with my drone and raspberry pi4, it turns off after 10 seconds when I connect the drone and e-port, what is the problem?

ChihYuanTeng, 2024-05-29:
@macrodk You should connect the drone and E-Port Development Kit, before you turn on the power of the M3T drone!

[image]

For more details, you can find in the following DJI Developer Website!
https://developer.dji.com/doc/payload-sdk-tutorial/en/quick-start/drone-port.html

macrodk, 2024-08-27:
thank you for your answer! But then we ran into another problem.

I keep getting errors like the one below
[image]

macrodk, 2024-08-27:
the required configuration code updates are
[image]
[image]
[image]
[image]

skshadan, 2024-09-30:
https://github.com/skshadan/Opencv-DJI-Drones.git