dji-sdk/Payload-SDK #159: Does Payload SDK provide supports for H20T features?

hnqiu, 2024-04-16:
Hi @dji-dev,
We're considering building applications with M350+H20T. Does the Payload SDK support H20T features, such as reading thermal data, pinpoint, smart check? There seems to be no such supports included in the list https://developer.dji.com/doc/payload-sdk-tutorial/en/basic-introduction/function-overview.html

Thanks,

dji-dev, 2024-04-17:
Agent comment from Leon in <a href='https://djisdksupport.zendesk.com/agent/tickets/105240'>Zendesk ticket #105240</a>:<br><br>Dear hnqiu,

Greetings! We appreciate your contact with DJI - Da-Jiang Innovations.

Unfortunately, the Precision Positioning and Temperature Data Acquisition functionalities you mentioned are not supported by the PSDK for H20T. You can refer to the supported functionalities at the following link:

https://developer.dji.com/doc/payload-sdk-tutorial/en/function-set/basic-function/basic-camera-management.html

Thank you for your support of DJI products! Wishing you all the best!

Best Regards,

DJI - Da-Jiang Innovations SDK Technical Support
<br><p><em><small>°°°</small></em></p>

hnqiu, 2024-04-18:
Thanks @dji-dev. That's quite helpful.

Another thing I'd like to know is I've seen video streams with wind estimates (wind speed, direction), but I couldn't find such info from the SDK. Is this supported? How is the wind status estimated?

dji-dev, 2024-04-19:
Agent comment from Leon in <a href='https://djisdksupport.zendesk.com/agent/tickets/105240'>Zendesk ticket #105240</a>:<br><br>Dear hnqiu,

Greetings! We appreciate your contact with DJI - Da-Jiang Innovations.

You can retrieve the wind speed level through this TOPIC, but the specific wind speed cannot be obtained from DJI_FC_SUBSCRIPTION_TOPIC_FLIGHT_ANOMALY. The threshold for level one is 12m/s, and anything greater than 12m/s is considered level two.

Thank you for your support of DJI products! Wishing you all the best!

Best Regards,

DJI - Da-Jiang Innovations SDK Technical Support
<br><p><em><small>°°°</small></em></p>

closed 2024-05-09 by hnqiu