dji-sdk/Payload-SDK #160: Arm-none-eabi-gcc

simon88, 2024-04-17:
Where is the lib for arm-none-eabi-gcc compiler? Is it remove? I can't find it in psdk lib

dji-dev, 2024-04-18:
Agent comment from Leon in <a href='https://djisdksupport.zendesk.com/agent/tickets/105347'>Zendesk ticket #105347</a>:<br><br>Dear simon88,

Greetings. We appreciate your contact with DJI - Da-Jiang Innovations.

We apologize for the delay. Regarding the static library, in order to streamline the PSDK code, we have moved it to a community shared drive. You can download it from the following link:
https://pan-sec.djicorp.com/s/tkYL4ERKD9k836s

Thank you for your support of DJI products! We wish you all the best.

Best Regards,

DJI - Da-Jiang Innovations SDK Technical Support
<br><p><em><small>°°°</small></em></p>

simon88, 2024-05-23:
ok nice thanks @dji-dev  and do you think the community version will be on github or nont ?

closed 2024-05-24 by simon88