dji-sdk/Payload-SDK #162: 树莓派4b

xdw-123, 2024-05-15:


xdw-123, 2024-05-15:
运行./dji_sdk_demo_linux_cxx示例程序，出现下面报错问题，而且示例和教程中的案例数量不相等，是我的什么配置没有弄好吗？希望尽快回复
[image]

dji-dev, 2024-05-16:
Agent comment from Leon in <a href='https://djisdksupport.zendesk.com/agent/tickets/107438'>Zendesk ticket #107438</a>:<br><br>尊敬的开发者，

您好，感谢您联系DJI 大疆创新。

这个报错是正常的现象，拓展口不支持云台功能，云台功能需要运行在负载口才能正常初始化。另外，目前PSDK的版本已经将所有功能重新归类，您看到的是分好类的选项，进入相关的类别以后才会展开功能。

感谢您对大疆产品的支持！祝您一切顺利！

Best Regards,
DJI 大疆创新SDK技术支持
<br><p><em><small>°°°</small></em></p>

xdw-123, 2024-05-17:
请问我想获取到在PDSK上得到无人机遥控器上点击已经有的按钮事件，可有推荐的案例？比如图片中的这个按键事件，相关案例太少也有可能我没找对地方，希望您这边有指导
[image]

dji-dev, 2024-05-17:
Agent comment from Leon in <a href='https://djisdksupport.zendesk.com/agent/tickets/107438'>Zendesk ticket #107438</a>:<br><br>尊敬的开发者，

您好，感谢您联系DJI 大疆创新。

您可以使用DjiTestWidget_GetWidgetValue函数来拿到控件的值，打印的时候会标注控件的index信息。

感谢您对大疆产品的支持！祝您一切顺利！

Best Regards,
DJI 大疆创新SDK技术支持
<br><p><em><small>°°°</small></em></p>