dji-sdk/Payload-SDK #165: Queries regarding Serial communication

kkishore9891, 2024-05-17:
Hello! We are using a payload computer that is powered by the Matrice M30T and and also communicates with the drone using UART for PSDK application using the same cable. The pinout is provided here:

[image]

When I power the drone external using a power supply instead of using the drone everything boots fine but when I boot using the drone. Sometimes the payload computer goes to the setup menu and the keyboard functionalities don't work as shown below:

[image]

I would like to know if the DJI drones sends data over UART even before the PSDK initialises. If so, is there a way to ensure that the drone sends data only when requested by the Payload SDK.

dji-dev, 2024-05-20:
Agent comment from Leon in <a href='https://djisdksupport.zendesk.com/agent/tickets/107633'>Zendesk ticket #107633</a>:<br><br>Dear kkishore9891,

Greetings. Thank you for reaching out to DJI - Da-Jiang Innovations.

We apologize for the delay and regret to inform you that the M30 series drones start outputting serial data immediately after powering on, and this behavior cannot be adjusted. We have observed that if you use an external power source, this issue does not occur. We suspect that the drone may not be providing a stable power supply to your load, causing the abnormal behavior. This could be due to voltage fluctuations when the drone is powered on.

We appreciate your support for DJI products! Wishing you all the best!

Best Regards,

DJI - Da-Jiang Innovations SDK Technical Support
<br><p><em><small>°°°</small></em></p>

kkishore9891, 2024-05-22:
Hello! Thank you for the reply! I would also like to know if the issue exists with the M3 drone as well. The one which is compatible with the DJI Dock 2.

dji-dev, 2024-05-22:
Agent comment from Leon in <a href='https://djisdksupport.zendesk.com/agent/tickets/107633'>Zendesk ticket #107633</a>:<br><br>Dear kkishore9891,

Greetings. We appreciate your inquiry with DJI Innovations.

Both M3E/T and M3D/TD models have encountered this issue.

Thank you for your support of DJI products! Wishing you all the best!

Best Regards,

DJI Innovations SDK Technical Support
<br><p><em><small>°°°</small></em></p>