dji-sdk/Payload-SDK #169: Support for nvidia_jetson?

kkishore9891, 2024-05-23:
Hello, I am using the Payload SDK version 3.8 for our Payload implementation and we have noticed that the code base for the nvidia_jetson platform is outdated and only the manifold2 platform is being constantly updated by the devs. I would like to know if running the manifold2 codebase on our Orin NX board would cause any issues or if it should switch to using the nvidia_jetson platform. What is the difference between using the code for nvidia_jetson platform and the manifold platform. Some information regarding these 2 code bases would be appreciated. We also noticed that the sample code doesn't compile for the nvidia_jetson platform.

dji-dev, 2024-05-24:
Agent comment from Leon in <a href='https://djisdksupport.zendesk.com/agent/tickets/108098'>Zendesk ticket #108098</a>:<br><br>Dear kkishore9891,

I hope this message finds you well. Thank you for reaching out to DJI - Da-Jiang Innovations.

We would like to inform you that there are no issues with using the Manifold 2 for developing code repositories. The different code repositories you mentioned are essentially the same, with distinctions only in the folder names. You can confidently proceed with your development.

Thank you for your support of DJI products! Wishing you continued success.

Best Regards,

DJI - Da-Jiang Innovations SDK Technical Support
<br><p><em><small>°°°</small></em></p>

kkishore9891, 2024-05-28:
When I tried to compile release 3.8 for nvidia jetson I got compilation errors and had to edit the HMS manager code a bit to make the code compile and even after getting that to work, the new widget functionalities were missing. It definitely is an older version of the stack.

kkishore9891, 2024-05-28:
I assure you that both the code repositories are not identical. Please confirm this with your team. Also, given the two different code bases, I would like to know what is the recommended code base version for a Nvidia Jetson Orin NX/Xavier NX board. We are facing some issues with our ROS2 wrapper collapsing with the Manifold2 build but seems to work after making some refinements in the nvidia build but the nvidia build seems to not be updated with the latest features.

dji-dev, 2024-05-28:
Agent comment from Leon in <a href='https://djisdksupport.zendesk.com/agent/tickets/108098'>Zendesk ticket #108098</a>:<br><br>Dear kkishore9891,

Greetings. We appreciate your contact with DJI - Da-Jiang Innovations.

We regret any inconvenience caused and would like to inform you that after further confirmation with our development team, the only platform currently being continuously updated is the Manifold2. Therefore, we recommend utilizing it when using Jetson Nano or other LINUX platforms.

Thank you for your support of DJI products! We wish you all the best!

Best Regards,

DJI - Da-Jiang Innovations SDK Technical Support
<br><p><em><small>°°°</small></em></p>

bonchan, 2024-06-19:
@dji-dev is manifold2 still a thing?