dji-sdk/Payload-SDK #171: waypoint mission resume information

kkishore9891, 2024-05-27:
Hello, I have a query regarding the waypoint mission v3.

Today I went out for a flight with my Matrice M30T drone using the RC. I was able to create a mission using the RC and execute it, pause it by pressing the pause button on the RC, fly to another point using the joystick and then resume the mission again using the RC. While doing so, the RC asked me if I wanted to resume from the breakpoint or continue to the next waypoint. Let us call this as the RC waypoint mission behaviour.

Now I would like to know if this is exactly how the waypoint mission v3 behaves as well. If I upload the kml file and execute the start mission action, I know the drone would simply execute the mission and if there is a RTH action in the KML file after the drone reaches the last waypoint, it would simply return to the home point position. This much is clear and understandable. But I am not sure about the pause and resume functionalities. The documentation is not clear regarding this. If I execute the pause action to interrupt the mission via the PSDK's function, would this behave like pressing the orange pause button on the RC? Can I fly using the joystick if the PSDK executed the pause action? Can I also pause a PSDK waypoint mission using the RC and resume the mission using the RC once again? If the resume action was executed using the PSDK instead, would it go back to the breakpoint or would it proceed to the next waypoint? 

To put it in simpler words I would like to know if it is possible to move away from the break point resulting from a pause action issued by the PSDK. If possible then can it only be done manually using the RC or if it can be done using the flight control feature of the PSDK as well: https://developer.dji.com/doc/payload-sdk-tutorial/en/function-set/basic-function/flight-control.html. And can we resume from a position which is away from the breakpoint? If so, would the drone proceed back to the breakpoint or would it proceed to the next waypoint in the mission?

dji-dev, 2024-05-28:
Agent comment from Leon in <a href='https://djisdksupport.zendesk.com/agent/tickets/108351'>Zendesk ticket #108351</a>:<br><br>Dear Developer,

Greetings and thank you for reaching out to DJI - Da-Jiang Innovations.

We appreciate your patience. Regarding your inquiry about the waypoint resume feature, we regret to inform you that this feature is only supported on the RC (Remote Controller). Unfortunately, this behavior is not supported in the PSDK (Payload SDK). If you wish to achieve waypoint resume functionality on the PSDK, you will need to save the remaining waypoint information before pausing the flight control. Once you have completed other tasks and return to the original position, you can then re-upload the remaining flight information.

We appreciate your support for DJI products! Wishing you every success!

Best Regards,

DJI - Da-Jiang Innovations SDK Technical Support

* * *
<br><p><em><small>°°°</small></em></p>

kkishore9891, 2024-05-29:
Thank you so much! We confirmed this behaviour today. We will follow your suggested approach.

kkishore9891, 2024-05-29:
Closing this issue.

kkishore9891, 2024-06-10:
Hello! I would like to know if you have any tutorial on how to edit KMZ files based on the current waypoint information  using C++ or python.