dji-sdk/Payload-SDK #175: M350 does not set the baudrate properly

stevedanomodolor, 2024-06-13:
To fly the drone, we set the user information using the set_user_info function as shown [here](https://github.com/umdlife/psdk_ros2/blob/34055786b810ace09182bab65b3f29301799533e/psdk_wrapper/src/psdk_wrapper.cpp#L137).
The specific implementation of the funtion can be found here.https://github.com/umdlife/psdk_ros2/blob/34055786b810ace09182bab65b3f29301799533e/psdk_wrapper/src/psdk_wrapper.cpp#L501. 

However, when using the M350, we noticed that the baud rate is not set correctly and is always fixed at 230400. This issue causes a problem when running both the camera and serial, leading to a drop in the data reception rate from the drone in some topics. I tried setting it to multiple values, even lower, but still its still fixed at 230400. 
```
stty -F /dev/dji_serial
speed 230400 baud; line = 0;
min = 0; time = 0;
-brkint -icrnl -imaxbel
-opost
-isig -icanon -echo -echoe
```

This does not happen in the m300. 

Is there a specific way to read the data when using the M350?

For more information, we are using the UART and network configuration, and the drone firmware version is v09.01.0102.

dji-dev, 2024-06-14:
Agent comment from Leon in <a href='https://djisdksupport.zendesk.com/agent/tickets/109682'>Zendesk ticket #109682</a>:<br><br>Dear stevedanomodolor,

Greetings. We appreciate your contact with DJI - Da-Jiang Innovations.

The website you provided does not offer the official ROS2 version from DJI. Therefore, we are unable to confirm if there are any issues with the interfaces. If you are unable to set the baud rate in the PSDK program, you can also configure it in the Payload SDK section of the DJI Assistant2 software.

Thank you for your support of DJI products! Wishing you every success.

Best Regards,

DJI - Da-Jiang Innovations SDK Technical Support
<br><p><em><small>°°°</small></em></p>

stevedanomodolor, 2024-06-14:
This is not possible with the m350 though. Its only possible with the m300 to change the baudrate through dji assistant. Can you send me instructions on how to do this?

The software psdk_ros2 uses the payload sdk and calls function directly from the payload sdk.

stevedanomodolor, 2024-06-14:
> Agent comment from Leon in <a href='https://djisdksupport.zendesk.com/agent/tickets/109682'>Zendesk ticket #109682</a>:<br><br>Dear stevedanomodolor,
>
> Greetings. We appreciate your contact with DJI [...]

Is there a difference in the way the baudrate is set for the m300 compared to the m350? ,because the m300 works fine with the same interface.

stevedanomodolor, 2024-06-14:
I was able to set it but through the onboard sdk. am i correct that the baudrate that has priority is the one that is set on the dji assistant right?

dji-dev, 2024-06-17:
Agent comment from Leon in <a href='https://djisdksupport.zendesk.com/agent/tickets/109682'>Zendesk ticket #109682</a>:<br><br>Dear stevedanomodolor,

Greetings. Thank you for reaching out to DJI Innovations.

We appreciate your patience. Regarding the baud rate, the part configured in the DA2 software takes precedence. If the DA2 software is configured with an unsupported baud rate, it will then retrieve the baud rate configured in the PSDK. If that is also not compatible, the baud rate will be automatically selected.

Thank you for your support of DJI products! Wishing you all the best!

Best Regards,

DJI Innovations SDK Technical Support
<br><p><em><small>°°°</small></em></p>

closed 2024-06-18 by stevedanomodolor