dji-sdk/Payload-SDK #176: Editing Waypoint Files

kkishore9891, 2024-06-13:
Hello, I am trying to edit waypoint files in real time during flight so that I can checkpoint a particular mission and resume it later from half the way as suggested here: https://github.com/dji-sdk/Payload-SDK/issues/171

Let's say I have a waypoint mission KMZ file 1 with its wpml file being:

`<?xml version="1.0" encoding="UTF-8"?>
<kml xmlns="http://www.opengis.net/kml/2.2" xmlns:wpml="http://www.dji.com/wpmz/1.0.5">
  <Document>
    <wpml:missionConfig>
      <wpml:flyToWaylineMode>safely</wpml:flyToWaylineMode>
      <wpml:finishAction>goHome</wpml:finishAction>
      <wpml:exitOnRCLost>executeLostAction</wpml:exitOnRCLost>
      <wpml:executeRCLostAction>goBack</wpml:executeRCLostAction>
      <wpml:takeOffSecurityHeight>20</wpml:takeOffSecurityHeight>
      <wpml:globalTransitionalSpeed>6.70570850372314</wpml:globalTransitionalSpeed>
      <wpml:droneInfo>
        <wpml:droneEnumValue>67</wpml:droneEnumValue>
        <wpml:droneSubEnumValue>1</wpml:droneSubEnumValue>
      </wpml:droneInfo>
      <wpml:payloadInfo>
        <wpml:payloadEnumValue>53</wpml:payloadEnumValue>
        <wpml:payloadSubEnumValue>0</wpml:payloadSubEnumValue>
        <wpml:payloadPositionIndex>0</wpml:payloadPositionIndex>
      </wpml:payloadInfo>
    </wpml:missionConfig>
    <Folder>
      <wpml:templateId>0</wpml:templateId>
      <wpml:executeHeightMode>relativeToStartPoint</wpml:executeHeightMode>
      <wpml:waylineId>0</wpml:waylineId>
      <wpml:distance>63.4997634887695</wpml:distance>
      <wpml:duration>46.82470703125</wpml:duration>
      <wpml:autoFlightSpeed>2.23523616790771</wpml:autoFlightSpeed>
      <Placemark>
        <Point>
          <coordinates>
            18.0648833227148,59.3520630245224
          </coordinates>
        </Point>
        <wpml:index>0</wpml:index>
        <wpml:executeHeight>57.3335762023926</wpml:executeHeight>
        <wpml:waypointSpeed>2.23523616790771</wpml:waypointSpeed>
        <wpml:waypointHeadingParam>
          <wpml:waypointHeadingMode>followWayline</wpml:waypointHeadingMode>
          <wpml:waypointHeadingAngle>0</wpml:waypointHeadingAngle>
          <wpml:waypointPoiPoint>0.000000,0.000000,0.000000</wpml:waypointPoiPoint>
          <wpml:waypointHeadingAngleEnable>0</wpml:waypointHeadingAngleEnable>
          <wpml:waypointHeadingPoiIndex>0</wpml:waypointHeadingPoiIndex>
        </wpml:waypointHeadingParam>
        <wpml:waypointTurnParam>
          <wpml:waypointTurnMode>toPointAndStopWithDiscontinuityCurvature</wpml:waypointTurnMode>
          <wpml:waypointTurnDampingDist>0</wpml:waypointTurnDampingDist>
        </wpml:waypointTurnParam>
        <wpml:useStraightLine>1</wpml:useStraightLine>
        <wpml:waypointGimbalHeadingParam>
          <wpml:waypointGimbalPitchAngle>0</wpml:waypointGimbalPitchAngle>
          <wpml:waypointGimbalYawAngle>0</wpml:waypointGimbalYawAngle>
        </wpml:waypointGimbalHeadingParam>
        <wpml:isRisky>0</wpml:isRisky>
      </Placemark>
      <Placemark>
        <Point>
          <coordinates>
            18.0648711087943,59.3519900857892
          </coordinates>
        </Point>
        <wpml:index>1</wpml:index>
        <wpml:executeHeight>57.3335762023926</wpml:executeHeight>
        <wpml:waypointSpeed>2.23523616790771</wpml:waypointSpeed>
        <wpml:waypointHeadingParam>
          <wpml:waypointHeadingMode>followWayline</wpml:waypointHeadingMode>
          <wpml:waypointHeadingAngle>0</wpml:waypointHeadingAngle>
[... 306 lines cut ...]
          <wpml:waypointHeadingPoiIndex>0</wpml:waypointHeadingPoiIndex>
        </wpml:waypointHeadingParam>
        <wpml:waypointTurnParam>
          <wpml:waypointTurnMode>toPointAndStopWithDiscontinuityCurvature</wpml:waypointTurnMode>
          <wpml:waypointTurnDampingDist>0</wpml:waypointTurnDampingDist>
        </wpml:waypointTurnParam>
        <wpml:useStraightLine>1</wpml:useStraightLine>
        <wpml:waypointGimbalHeadingParam>
          <wpml:waypointGimbalPitchAngle>0</wpml:waypointGimbalPitchAngle>
          <wpml:waypointGimbalYawAngle>0</wpml:waypointGimbalYawAngle>
        </wpml:waypointGimbalHeadingParam>
        <wpml:isRisky>0</wpml:isRisky>
      </Placemark>
    </Folder>
  </Document>
</kml>
`

I did manage to write a python script which is capable of getting a KMZ file and the waypoint index at which I can clip the mission to generate a new KMZ file. The problem I had howevever is that I am not sure how to calculate the values:

`
      <wpml:distance>27.4170398712158</wpml:distance>
      <wpml:duration>20.1444892883301</wpml:duration>
`
I am assuming the RC uses some kind of logic to compute those values. I would also like to know if the drone would fly properly if those two values were wrong.

dji-dev, 2024-06-14:
Agent comment from Leon in <a href='https://djisdksupport.zendesk.com/agent/tickets/109691'>Zendesk ticket #109691</a>:<br><br>Dear kkishore9891,

Greetings! We appreciate your contact with DJI - Da-Jiang Innovations.

May we inquire about the specific SDK you are using? We apologize for the inconvenience, but we may need to confirm this information first before we can assist you more effectively.

Thank you for your support of DJI products! Wishing you all the best!

Best Regards,

DJI - Da-Jiang Innovations SDK Technical Support
<br><p><em><small>°°°</small></em></p>

kkishore9891, 2024-06-17:
I am using the Payload SDK for the Matrice M30T drone.

dji-dev, 2024-06-18:
Agent comment from Leon in <a href='https://djisdksupport.zendesk.com/agent/tickets/109691'>Zendesk ticket #109691</a>:<br><br>Dear kkishore9891,

Greetings. Thank you for reaching out to DJI - Da-Jiang Innovations.

We appreciate your patience. We have confirmed that these two values will affect the execution of your flight mission, and these values are generated by the RC.

If you wish to resume the waypoint mission midway,

You can refer to the callback in the V3 waypoint mission to determine which waypoint index has been executed. Then, using a script, locate this index in the V3 waypoint file. You can delete the previously executed waypoints and reset the waypoints following the index+1 to 0. Subsequent waypoints will start from 0, 1, 2, and so on. This way, you can retain the original waypoint information without the need for recalculating.

Thank you for your support of DJI products! Wishing you every success!

Best Regards,

DJI - Da-Jiang Innovations SDK Technical Support
<br><p><em><small>°°°</small></em></p>