dji-sdk/Payload-SDK #178: remove negotiated device sync error

akgongon, 2024-06-17:
I recieve the below error message with PSDK 3.9.0

PSDK3.9.0 +  + Matrice350RTK + arm-linux-gnueabihf-gcc

[0.013][core]-[Info]-[DjiCore_Init:106) Payload SDK Version : V3.9.0-beta.0-build.2044
[0.382][adapter]-[Info]-[DjiAccessAdapter_Init:231) Identify aircraft series is Matrice 350 Series
[0.382][adapter]-[Info]-[DjiAccessAdapter_Init:264) Identify mount position type is Payload Port Type
[0.386][adapter]-[Info]-[DjiAccessAdapter_Init:365) Identity uart0 baudrate is 460800 bps
[0.389][adapter]-[Info]-[DjiPayloadNegotiate_Init:243) Waiting payload negotiate finished.
[1.425][core]-[Info]-[DjiIdentityVerify_UpdatePolicy:474) Updating dji sdk policy file...
[2.425][core]-[Info]-[DjiIdentityVerify_UpdatePolicy:482) Update dji sdk policy file successfully
[3.452][infor]-[Info]-[DjiAircraftInfo_GetAdapterSerialNumber:806) Get serial number of Skyport V2 or X-Port is 
[3.458][core]-[Info]-[DjiCore_Init:174) Identify AircraftType = Matrice 350 RTK, MountPosition = Payload Port NO1, SdkAdapterType = Skyport V2
[3.481][time_sync]-[Info]-[DjiTimeSync_PushAppTimeHandle:111) Current time: 2024.06.17 21:00:57.
[12.532][adapter]-[Info]-[DjiPayloadNegotiate_Init:243) Waiting payload negotiate finished.
[13.532][adapter]-[Info]-[DjiPayloadNegotiate_Init:243) Waiting payload negotiate finished.
[14.532][adapter]-[Info]-[DjiPayloadNegotiate_Init:243) Waiting payload negotiate finished.
[15.532][adapter]-[Info]-[DjiPayloadNegotiate_Init:243) Waiting payload negotiate finished.
[16.533][adapter]-[Info]-[DjiPayloadNegotiate_Init:243) Waiting payload negotiate finished.
**[21.555][cam]-[Error]-[DjiPayloadCamera_SetVideoStreamType:799) remove negotiated device sync error: 0x000000E1.
[21.555][user]-[Error]-[DjiTest_CameraEmuBaseStartService:1433) DJI camera set video stream error.
[21.555][user]-[Error]-[main:311) camera emu common init error**
[22.301][core]-[Info]-[DjiCore_ApplicationStart:238) Start dji sdk application
[737.476][infor]-[Info]-[DjiAircraftInfo_NotifyMobileAppInfoHandle:715) Set mobile app info, language is Japanese, screen type is Big Screen
[739.668][downloader]-[Warn]-[DjiDataDownloader_SendDataUpdate:559) Camera Get App File Send Ack TimeOut

My code has
`s_cameraVideoStreamType = DJI_CAMERA_VIDEO_STREAM_TYPE_H264_CUSTOM_FORMAT;`

I'm wondering that it have something relation with this error.

dji-dev, 2024-06-18:
Agent comment from Leon in <a href='https://djisdksupport.zendesk.com/agent/tickets/109921'>Zendesk ticket #109921</a>:<br><br>Dear akgongon,

Greetings. We appreciate your contact with DJI - Da-Jiang Innovations.

We regret any inconvenience caused. We have taken note of the issue you mentioned, and it is currently being addressed. Could you please inform us of any impact on your end when switching to DJI-H264 from s_cameraVideoStreamType set as Custom-H264? For now, you may switch to DJI-H264 or use version 3.8.1 for development purposes as we are actively working on fixing this issue.

Thank you for your support of DJI products! Wishing you continued success!

Best Regards,

DJI - Da-Jiang Innovations SDK Technical Support
<br><p><em><small>°°°</small></em></p>

akgongon, 2024-06-19:
Thank you for your reply.

My payload needs to monitor the camera in real-time.
The H264_CUSTOM_FORMAT method is much familiar than DJI-H264 way in my project.
My project has another process to send the video stream with gstreamer.

I strongly hope that the wizards in DJI will fix this problem soon✨
I back to 3.8.1 until the problem will be fixed.

dji-dev, 2024-06-19:
Agent comment from Leon in <a href='https://djisdksupport.zendesk.com/agent/tickets/109921'>Zendesk ticket #109921</a>:<br><br>Dear akgongon,

Greetings. Thank you for reaching out to DJI - Da-Jiang Innovations.

Upon reviewing your situation, we regret any inconvenience caused. We have escalated this matter to our development team for prompt analysis. We apologize once again.

We appreciate your support for DJI products! Wishing you all the best!

Best Regards,

DJI - Da-Jiang Innovations SDK Technical Support
<br><p><em><small>°°°</small></em></p>

akgongon, 2024-07-19:
Is not this issue solved yet in 3.9.1 ?
I try the new 3.9.1 but I recieve same error message.

tuyaliang, 2024-07-22:
Same error.

SJJ-dot, 2025-02-20:
3.9.2 还是没有解决。