dji-sdk/Payload-SDK #180: video stream not working in M3TD

kkishore9891, 2024-07-03:
Hello! I Tried to run the sample code in M3TD after setting up USB bulk configuration. I found that the image stream is not being displayed when running the sample script as shown in the video below.

https://github.com/dji-sdk/Payload-SDK/assets/34810513/ffe16347-5543-468e-ac70-89255a778c8c

I would like to know if I need to modify anything in the sample code to recieve the image stream. 

For example in my separate code base inspired from the sample test_liveview.cpp code I used a setup like this for the Matrice M30T drone as shown below:

```bash
T_DjiReturnCode LiveviewSample::StartWideCameraStream(CameraImageCallback callback, void *userData)
{
    // Obtain the decoder for Wide camera. Perhaps it will work with any decoder. Need to test.
    auto decoder = streamDecoder.find(DJI_LIVEVIEW_CAMERA_POSITION_NO_1);
    // USER_LOG_ERROR("Starting wide");
    
    // If the decoder exists, start it and associate it with a callback
    if ((decoder != streamDecoder.end()) && decoder->second)
    {
        decoder->second->init();
        decoder->second->registerCallback(callback, userData);

        return DjiLiveview_StartH264Stream(DJI_LIVEVIEW_CAMERA_POSITION_NO_1, DJI_LIVEVIEW_CAMERA_SOURCE_M30T_WIDE,
                                           LiveviewConvertH264ToRgbCallback);
    }
    // Error handling
    else
    {
        return DJI_ERROR_SYSTEM_MODULE_CODE_NOT_FOUND;
    }
}

// Function that associates a callback with the thermal camera stream and a user defined data
// in our case the user defined data is a struct object with a publisher and clock.
T_DjiReturnCode LiveviewSample::StartThermalCameraStream(CameraImageCallback callback, void *userData)
{
    // Obtain the decoder for thermal camera. Perhaps it will work with any decoder. Need to test.
    auto decoder = streamDecoder.find(DJI_LIVEVIEW_CAMERA_POSITION_NO_1);

    // If the decoder exists, start it and associate it with a callback
    if ((decoder != streamDecoder.end()) && decoder->second)
    {
        decoder->second->init();
        decoder->second->registerCallback(callback, userData);
        USER_LOG_ERROR("Starting Thermal Camera Streaming");
        return DjiLiveview_StartH264Stream(DJI_LIVEVIEW_CAMERA_POSITION_NO_1, DJI_LIVEVIEW_CAMERA_SOURCE_M30T_IR,
                                           LiveviewConvertH264ToRgbCallback);
    }
    // Error handling
    else
    {
        return DJI_ERROR_SYSTEM_MODULE_CODE_NOT_FOUND;
    }
}
```
This worked for the M30T drone but using something similar like DJI_LIVEVIEW_CAMERA_POSITION_NO_1 and DJI_LIVEVIEW_CAMERA_SOURCE_M3TD_IR or DJI_LIVEVIEW_CAMERA_SOURCE_M3TD_VIS doesn't send any images to the callback function. I am not sure if USB bulk might have some issue. I would first like to get the video feed working in the sample script so that I can later use that to modify our code base to accomodate the M3TD drone.

dji-dev, 2024-07-04:
Agent comment from Leon in <a href='https://djisdksupport.zendesk.com/agent/tickets/111143'>Zendesk ticket #111143</a>:<br><br>Dear kkishore9891,

Greetings. We appreciate your contact with DJI Innovations.

Could you please test the DjiTest_LiveviewRunSample function in the test_liveview.c file located in payload-sdk\samples\sample_c\module_sample\liveview first? Afterwards, add print statements in either the DjiTest_FpvCameraStreamCallback or DjiTest_PayloadCameraStreamCallback callback functions to verify if the callback is being triggered. If the callback is not being triggered, it indicates that there may still be an issue with the bulk link.

Thank you for your support of DJI products! We wish you every success.

Best Regards,

DJI Innovations SDK Technical Support
<br><p><em><small>°°°</small></em></p>

kkishore9891, 2024-07-04:
Hello I will try this and let you know but I tried something else in the Matrice M30T with the same jetson setup that confirms that there is some issue with the bulk like.

 Since Matrice M30T support DJI_USE_UART_AND_NETWORK_DEVICE and DJI_USE_UART_AND_USB_BULK_DEVICE options I connected my hardware setup to M30T instead of M3TD and simply swapped DJI_USE_UART_AND_USB_BULK_DEVICE with DJI_USE_UART_AND_NETWORK_DEVICE, and the image subscription works. While the same M30T with the exact same code and hardware setup but using DJI_USE_UART_AND_USB_BULK_DEVICE instead in the dji_sdk_config.h causes the image stream not to work in the M30T. Since M3TD doesn't support DJI_USE_UART_AND_NETWORK_DEVICE I have never been able to get the image stream to work. But I will try what you said soon and update you.

kkishore9891, 2024-07-04:
Hello, so I tried to run the test_livew.c program's DjiTest_LiveviewRunSample() function as per your suggestion and added some print statements to the callback as shown in the script below:

```bash
/**
 ********************************************************************
 * @file    test_liveview.c
 * @brief
 *
 * @copyright (c) 2021 DJI. All rights reserved.
 *
 * All information contained herein is, and remains, the property of DJI.
 * The intellectual and technical concepts contained herein are proprietary
 * to DJI and may be covered by U.S. and foreign patents, patents in process,
 * and protected by trade secret or copyright law.  Dissemination of this
 * information, including but not limited to data and other proprietary
 * material(s) incorporated within the information, in any form, is strictly
 * prohibited without the express written consent of DJI.
 *
 * If you receive this source code without DJI’s authorization, you may not
 * further disseminate the information, and you must immediately remove the
 * source code and notify DJI of its removal. DJI reserves the right to pursue
 * legal actions against you for any loss(es) or damage(s) caused by your
 * failure to do so.
 *
 *********************************************************************
 */

/* Includes ------------------------------------------------------------------*/
#include <utils/util_misc.h>
#include <widget_interaction_test/test_widget_interaction.h>
#include "test_liveview.h"
#include "dji_liveview.h"
#include "dji_logger.h"
#include "dji_platform.h"
#include "dji_aircraft_info.h"
#include "time.h"

/* Private constants ---------------------------------------------------------*/
#define TEST_LIVEVIEW_STREAM_FILE_PATH_STR_MAX_SIZE             256
#define TEST_LIVEVIEW_STREAM_STROING_TIME_IN_SECONDS            20

#define TEST_LIVEVIEW_STREAM_REQUEST_I_FRAME_ON                 1
#define TEST_LIVEVIEW_STREAM_REQUEST_I_FRAME_TICK_IN_SECONDS    5

/* Private types -------------------------------------------------------------*/

/* Private values -------------------------------------------------------------*/
static char s_fpvCameraStreamFilePath[TEST_LIVEVIEW_STREAM_FILE_PATH_STR_MAX_SIZE];
static char s_payloadCameraStreamFilePath[TEST_LIVEVIEW_STREAM_FILE_PATH_STR_MAX_SIZE];

/* Private functions declaration ---------------------------------------------*/
static void DjiTest_FpvCameraStreamCallback(E_DjiLiveViewCameraPosition position, const uint8_t *buf,
                                            uint32_t bufLen);
static void DjiTest_PayloadCameraStreamCallback(E_DjiLiveViewCameraPosition position, const uint8_t *buf,
                                                uint32_t bufLen);

/* Exported functions definition ---------------------------------------------*/
T_DjiReturnCode DjiTest_LiveviewRunSample(E_DjiMountPosition mountPosition)
{
    T_DjiReturnCode returnCode;
    T_DjiOsalHandler *osalHandler = DjiPlatform_GetOsalHandler();
    time_t currentTime = time(NULL);
    struct tm *localTime = NULL;
    T_DjiAircraftInfoBaseInfo aircraftInfoBaseInfo = {0};

    USER_LOG_INFO("Liveview sample start");
    DjiTest_WidgetLogAppend("Liveview sample start");

    returnCode = DjiAircraftInfo_GetBaseInfo(&aircraftInfoBaseInfo);
    if (returnCode != DJI_ERROR_SYSTEM_MODULE_CODE_SUCCESS) {
[... 333 lines cut ...]
            break;
        case 'c':
            DjiTest_LiveviewRunSample(s_mountPosition);
            break;
        case 'd':
            DjiUser_RunStereoVisionViewSample();
            break;
        case 'e':
            DjiUser_RunCameraManagerSample();
            break;
        case 'f':
            returnCode = DjiTest_PositioningStartService();
            if (returnCode != DJI_ERROR_SYSTEM_MODULE_CODE_SUCCESS) {
                USER_LOG_ERROR("rtk positioning sample init error");
                break;
            }

            USER_LOG_INFO("Start rtk positioning sample successfully");
            break;
        default:
            break;
    }
```

Thus, it is confirmed now that the USB Bulk link has some flaw. Could you please look into the procedure that I have followed as explain in the previous comment to see if I made some mistake while setting up the USB Bulk link? The tutorial provided in PSDK documentation offers the suggestions for Jetson Nano but since I am using Orin Nano I had to make some changes as I have indicated with comments.

dji-dev, 2024-07-05:
Agent comment from Leon in <a href='https://djisdksupport.zendesk.com/agent/tickets/111143'>Zendesk ticket #111143</a>:<br><br>Dear Developer,

Greetings and thank you for reaching out to DJI Innovations.

We appreciate the steps and validation conclusions you provided. We have carefully reviewed your validation on the M30T model. Firstly, there is no necessary connection between BULK and NETWORK. The NETWORK link you used should be based on RNDIS for communication, which should not be affected by whether your BULK is configured correctly. Additionally, you mentioned that even when switching to BULK, the M30T still cannot retrieve data properly.

We also noticed that there is no information printed from the callback function in your log, indicating that your bulk or PSDK's hal_usb_bulk.h file is still not configured correctly.

Perhaps you can locate the nv-l4t-usb-device-mode-start.sh file in your development board and add the following code snippet at the end (please adjust the paths accordingly):

    enable_bulk=1 if [${enable_bulk} -eq 1]; then mkdir -p /dev/usb-ffs cfg_str="${cfg_str}+BULK1" mkdir -p /dev/usb-ffs/bulk1 func=functions/ffs.bulk1 mkdir -p "${func}" ln -sf "${func}" "${cfg}" mount -o mode=0777 -o uid=2000 -o gid=2000 -t functionfs bulk1 /dev/usb-ffs/bulk1 #/home/dji/Desktop/startup_bulk/startup_bulk /dev/usb-ffs/bulk1 & /home/sutpc/Downloads/UAV/usb-bulk-configuration-reference/startup_bulk /dev/usb-ffs/bulk1 & sleep 3 cfg_str="${cfg_str}+BULK2" mkdir -p /dev/usb-ffs/bulk2 func=functions/ffs.bulk2 mkdir -p "${func}" ln -sf "${func}" "${cfg}" mount -o mode=0777 -o uid=2000 -o gid=2000 -t functionfs bulk2 /dev/usb-ffs/bulk2 #/home/dji/Desktop/startup_bulk/startup_bulk /dev/usb-ffs/bulk2 & /home/sutpc/Downloads/UAV/usb-bulk-configuration-reference/startup_bulk /dev/usb-ffs/bulk2 & sleep 3 fi

Thank you for your support of DJI products! Wishing you every success!

Best Regards,

DJI Innovations SDK Technical Support
<br><p><em><small>°°°</small></em></p>

kkishore9891, 2024-07-05:
Not sure if this helps but I tried to run the compiled startup_bulk executable in the jetson and here is the output:

nvidia@nvidia:~/Desktop/nano-usb-config$ ./startup_bulk /dev/usb-ffs/bulk1 &
[1] 5552
unable do write descriptors
Unable to prepare ffs: -3
[1]+  Exit 253                ./startup_bulk /dev/usb-ffs/bulk1
nvidia@nvidia:~/Desktop/nano-usb-config$

dji-dev, 2024-07-08:
Agent comment from Leon in <a href='https://djisdksupport.zendesk.com/agent/tickets/111143'>Zendesk ticket #111143</a>:<br><br>Dear kkishore9891,

Greetings. We appreciate your contact with DJI Innovations.

Please incorporate the modifications provided in our previous response to the end of your current source file. Run your script with the updated path, and refrain from executing any other commands for now. Afterwards, you can observe any error messages that may appear.

You can download the script nv-l4t-usb-device-mode-start.sh here.

Thank you for your support of DJI products! We wish you every success.

Best Regards,

DJI Innovations SDK Technical Support
<br><p><em><small>°°°</small></em></p>

casper9429-kth, 2024-07-09:
> Agent comment from Leon in [Zendesk ticket #111143](https://djisdksupport.zendesk.com/agent/tickets/111143):Dear kkishore9891,
> 
> Greetings. We appreciate your contact with DJI Innovations.
> 
> Please incorporate the modifications provided in our previous response to the end of your current source file. Run your script with the updated path, and refrain from executing any other commands for now. Afterwards, you can observe any error messages that may appear.
> 
> You can download the script nv-l4t-usb-device-mode-start.sh here.
> 
> Thank you for your support of DJI products! We wish you every success.
> 
> Best Regards,
> 
> DJI Innovations SDK Technical Support
> 
> _°°°_

Hi dji-dev!

I am working on this problem together with @kkishore9891.

We added: 
```
enable_bulk=1
if [ ${enable_bulk} -eq 1 ]; then
    mkdir -p /dev/usb-ffs
    
    cfg_str="${cfg_str}+BULK1" 
    mkdir -p /dev/usb-ffs/bulk1
    func=functions/ffs.bulk1
    mkdir -p "${func}"
    ln -sf "${func}" "${cfg}"
    mount -o mode=0777 -o uid=2000 -o gid=2000 -t functionfs bulk1 /dev/usb-ffs/bulk1
    /home/nvidia/Desktop/nano-usb-config/startup_bulk /dev/usb-ffs/bulk1 &
    sleep 3

    cfg_str="${cfg_str}+BULK2" 
    mkdir -p /dev/usb-ffs/bulk2
    func=functions/ffs.bulk2
    mkdir -p "${func}"
    ln -sf "${func}" "${cfg}"
    mount -o mode=0777 -o uid=2000 -o gid=2000 -t functionfs bulk2 /dev/usb-ffs/bulk2
    /home/nvidia/Desktop/nano-usb-config/startup_bulk /dev/usb-ffs/bulk2 &
    sleep 3
fi
```
To the end of `/opt/nvidia/l4t-usb-device-mode/nv-l4t-usb-device-mode-start.sh`.

We get the following errors:
```
nvidia@nvidia:/opt/nvidia/l4t-usb-device-mode$ sudo journalctl -fu nv-l4t-usb-device-mode.service 
Jul 08 17:31:30 nvidia systemd[1]: Starting Configure USB flashing port for device mode...
Jan 01 01:00:29 nvidia nv-l4t-usb-device-mode-start.sh[547]: /
Jan 01 01:00:31 nvidia nv-l4t-usb-device-mode-start.sh[1199]: /sys/devices/platform/bus@0/3520000.padctl/usb2-0/usb_role/usb2-0-role-switch
Jan 01 01:00:31 nvidia nv-l4t-usb-device-mode-start.sh[1397]: ln: failed to create symbolic link 'configs/c.1': No such file or directory
Jan 01 01:00:31 nvidia systemd[1]: nv-l4t-usb-device-mode.service: Main process exited, code=exited, status=1/FAILURE
Jan 01 01:00:34 nvidia nv-l4t-usb-device-mode-stop.sh[1958]: rmdir: failed to remove 'l4t/functions/ecm.usb0/': No such file or directory
Jan 01 01:00:36 nvidia systemd[1]: nv-l4t-usb-device-mode.service: Failed with result 'exit-code'.
Jan 01 01:00:36 nvidia systemd[1]: Failed to start Configure USB flashing port for device mode.
```

We are really thankful for all the help!

**Background**
We are both working at the company FLOX robotics AB, and trying to make a jetson Orin to work together with the DJI M3TD through PSDK. And we are a bit on a tight deadline, if you over business to business support, please contact us on `casper@floxrobotics.com`.

Best, Casper

casper9429-kth, 2024-07-09:
Additional information @dji-dev . The usb_gadget c.1 and additional files gets created if we add the code you gave us to somewhere in the middle of the `/opt/nvidia/l4t-usb-device-mode/nv-l4t-usb-device-mode-start.sh` corresponding to the file ` psdk-usb-configure.sh` given in the guide [Jetson Nano Quick Start](https://developer.dji.com/doc/payload-sdk-tutorial/en/quick-start/quick-guide/jetson-nano.html). They do not get created if we add the line you gave us to the end off `nv-l4t-usb-device-mode-start.sh`.

When we connect the Jetson Orin to itself from the gadget usb to an arbitrary usb port on the same jetson orin we get this information when typing `lsusb -d 0955:7020 -v`

```
nvidia@nvidia:~$ lsusb -d 0955:7020 -v

Bus 001 Device 008: ID 0955:7020 NVIDIA Corp. L4T (Linux for Tegra) running on Tegra
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.10
  bDeviceClass          239 Miscellaneous Device
  bDeviceSubClass         2 
  bDeviceProtocol         1 Interface Association
  bMaxPacketSize0        64
  idVendor           0x0955 NVIDIA Corp.
  idProduct          0x7020 L4T (Linux for Tegra) running on Tegra
  bcdDevice            0.02
  iManufacturer           1 NVIDIA
  iProduct                2 Linux for Tegra
  iSerial                 3 1420524216468
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0127
    bNumInterfaces          9
    bConfigurationValue     1
    iConfiguration          4 
    bmAttributes         0xe0
      Self Powered
      Remote Wakeup
    MaxPower                2mA
    Interface Association:
      bLength                 8
      bDescriptorType        11
      bFirstInterface         0
      bInterfaceCount         2
      bFunctionClass          2 Communications
      bFunctionSubClass       6 Ethernet Networking
      bFunctionProtocol       0 
      iFunction               7 
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         2 Communications
      bInterfaceSubClass      2 Abstract (modem)
      bInterfaceProtocol    255 Vendor Specific (MSFT RNDIS?)
      iInterface              5 
      CDC Header:
        bcdCDC               1.10
      CDC Call Management:
        bmCapabilities       0x00
        bDataInterface          1
      CDC ACM:
        bmCapabilities       0x00
      CDC Union:
        bMasterInterface        0
        bSlaveInterface         1 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
[... 259 lines cut ...]
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x89  EP 9 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0000  1x 0 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x06  EP 6 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0000  1x 0 bytes
        bInterval               0
```

Thank you again for all the help!

Best, Casper

casper9429-kth, 2024-07-09:
Hi again!

We solved the problem!

Following this guide [树莓派4B配置USB device RNDIS 和 BULK
](https://sdk-forum.dji.net/hc/zh-cn/articles/10232604141465-%E6%A0%91%E8%8E%93%E6%B4%BE4B%E9%85%8D%E7%BD%AEUSB-device-RNDIS-%E5%92%8C-BULK) , we recompiled the `startup_bulk.c`
And added 
```
/******************** Little Endian Handling ********************************/

/*
* cpu_to_le16/32 are used when initializing structures, a context where a
* function call is not allowed. To solve this, we code cpu_to_le16/32 in a way
* that allows them to be used when initializing structures.
*/

#if __BYTE_ORDER == __LITTLE_ENDIAN
#define cpu_to_le16(x) (x)
#define cpu_to_le32(x) (x)
#else
#define cpu_to_le16(x) ((((x) >> 8) & 0xffu) | (((x) & 0xffu) << 8))
#define cpu_to_le32(x) \
((((x) & 0xff000000u) >> 24) | (((x) & 0x00ff0000u) >> 8) | \
(((x) & 0x0000ff00u) << 8) | (((x) & 0x000000ffu) << 24))
#endif

#define le32_to_cpu(x) le32toh(x)
#define le16_to_cpu(x) le16toh(x)
```

and replaced:
```
htole32  --> cpu_to_le32()

htole16 --> cpu_to_le16()
```

We also downgraded and install these dependencies:
```
/* Update software repository */
sudo apt-get update
sudo apt-get upgrade   
sudo apt-get install automake
sudo apt-get install libaio-dev
/* Download opus-1.3.1 source code from https://opus-codec.org/ and install */
tar -xzvf opus-1.3.1.tar.gz
cd opus-1.3.1/
autoreconf  -f  -i
./configure
make -j16
sudo make install
/* Download and install FFmpeg 4.3 source code from GitHub */
tar -zxvf ffmpeg-4.3.2.tar.gz
./configure --enable-shared
make && make install
/* Verify OpenCV installation */
opencv_version
/* Install libusb */
sudo apt-get install libusb-1.0-0-dev
```

We added the line @dji-dev gave us, but in the middle of the file corresponding to where it is in files given in the jetson quick start tutorial.

In PSDK, `hal_usb_bulk.h` we also used these settings.

```
#define LINUX_USB_BULK1_EP_OUT_FD               "/dev/usb-ffs/bulk1/ep1"
#define LINUX_USB_BULK1_EP_IN_FD                "/dev/usb-ffs/bulk1/ep2"

#define LINUX_USB_BULK1_INTERFACE_NUM           (7)
#define LINUX_USB_BULK1_END_POINT_IN            (0x88)
#define LINUX_USB_BULK1_END_POINT_OUT           (5)

#define LINUX_USB_BULK2_EP_OUT_FD               "/dev/usb-ffs/bulk2/ep1"
#define LINUX_USB_BULK2_EP_IN_FD                "/dev/usb-ffs/bulk2/ep2"

#define LINUX_USB_BULK2_INTERFACE_NUM           (8)
#define LINUX_USB_BULK2_END_POINT_IN            (0x89)
#define LINUX_USB_BULK2_END_POINT_OUT           (6)
```

And now it works!

dji-dev, 2024-07-10:
Agent comment from Leon in <a href='https://djisdksupport.zendesk.com/agent/tickets/111143'>Zendesk ticket #111143</a>:<br><br>Dear kkishore9891,

I hope this message finds you well. Thank you for reaching out to DJI - Da-Jiang Innovations.

We apologize for the delayed response due to the high volume of inquiries recently.

We are pleased to hear that your issue has been resolved. It appears that you have successfully resolved it by referring to an article in our knowledge base. We have noticed that you have also downgraded some dependencies. Could you please confirm if these actions were necessary? We can supplement our documentation to provide developers with more references.

Thank you for your support of DJI products! Wishing you continued success.

Best Regards,

DJI - Da-Jiang Innovations SDK Technical Support
<br><p><em><small>°°°</small></em></p>

sehadnassim, 2024-10-18:
I am also encountring this issue but through ethernet cable. Can you confirm that it works through network. I am trying o pipe the packets to udp and play with ffplay. But i get : 

```
SEI type 1 size 1440 truncated at 1152f=0/0   
[h264 @ 0x55c3d8bc20] SEI type 1 size 1440 truncated at 1152itrate=1024.5kbits/s speed=0.552x    
[h264 @ 0x55c3d75b30] SEI type 1 size 1440 truncated at 1152f=0/0   
[h264 @ 0x55c3e379b0] SEI type 1 size 1440 truncated at 1152f=0/0   
av_interleaved_write_frame(): Broken pipe=    0KB sq=    0B f=0/0 
```