dji-sdk/Payload-SDK #182: Does the PSDK support transferring and storing the photos captured by the drone to the onboard computer?

lcyyyao, 2024-07-05:
Does the PSDK support transferring and storing the photos captured by the drone to the onboard computer?

dji-dev, 2024-07-05:
Agent comment from Leon in <a href='https://djisdksupport.zendesk.com/agent/tickets/111293'>Zendesk ticket #111293</a>:<br><br>Dear lcyyyao,

Greetings. Thank you for reaching out to DJI - Da-Jiang Innovations.

If you are using FPV for drone shooting, it does not support saving. However, if you are using DJI's official payload camera for shooting, you can store the footage in the onboard computer. For example, with H20T, you can use DjiTest_CameraManagerMediaDownloadAndDeleteMediaFile to download the files to the onboard computer directory.

We appreciate your support for DJI products! Wishing you all the best!

Best Regards,

DJI - Da-Jiang Innovations SDK Technical Support
<br><p><em><small>°°°</small></em></p>

lcyyyao, 2024-08-05:
Dear DJI SDK Technical Support Team,

Greetings. Thank you for your prompt response and the information provided regarding the FPV and official payload camera functionalities.

I have noted that the DjiTest_CameraManagerMediaDownloadAndDeleteMediaFile function is mentioned only in the dji-sdk/Payload-SDK · samples/sample_c/module_sample/camera_manager/test_camera_manager.c file.

I would like to inquire if there is any available C++ code that can be used to achieve the same functionality—specifically, to read photos taken by the H20 sensor and transfer them to the onboard computer for storage.

Your assistance and support in this matter would be greatly appreciated.

Thank you and best regards,
lcyyyao

lcyyyao, 2024-08-05:
> Agent comment from Leon in [Zendesk ticket #111293](https://djisdksupport.zendesk.com/agent/tickets/111293):Dear lcyyyao,
> 
> Greetings. Thank you for reaching out to DJI - Da-Jiang Innovations.
> 
> If you are using FPV for drone shooting, it does not support saving. However, if you are using DJI's official payload camera for shooting, you can store the footage in the onboard computer. For example, with H20T, you can use DjiTest_CameraManagerMediaDownloadAndDeleteMediaFile to download the files to the onboard computer directory.
> 
> We appreciate your support for DJI products! Wishing you all the best!
> 
> Best Regards,
> 
> DJI - Da-Jiang Innovations SDK Technical Support
> 
> _°°°_

Dear DJI SDK Technical Support Team,

Greetings. Thank you for your prompt response and the information provided regarding the FPV and official payload camera functionalities.

I have noted that the DjiTest_CameraManagerMediaDownloadAndDeleteMediaFile function is mentioned only in the dji-sdk/Payload-SDK · samples/sample_c/module_sample/camera_manager/test_camera_manager.c file.

I would like to inquire if there is any available C++ code that can be used to achieve the same functionality—specifically, to read photos taken by the H20 sensor and transfer them to the onboard computer for storage.

Your assistance and support in this matter would be greatly appreciated.

Thank you and best regards,
lcyyyao