dji-sdk/Payload-SDK #183: M30T跑psdk3.8.0报错

sthkklog, 2024-07-05:
#define CONFIG_HARDWARE_CONNECTION         DJI_USE_UART_AND_NETWORK_DEVICE
使用串口接受IMU等信息，psdk基本能成功；但是改成DJI_USE_UART_AND_NETWORK_DEVICE后，通过网口传图像psdk基本都报错，只是偶尔成功，下面是报错信息：
[image]

[754.989][adapter]-[Info]-[DjiPayloadNegotiate_Init:165) Waiting for negotiator to be ready... 
[756.990][linker]-[Warn]-[DjiCommand_SendAsyncHandle:894) Command async send retry: index = 0, retryTimes = 3, cmdSet = 0, cmdId = 0 
[757.991][linker]-[Warn]-[DjiCommand_SendAsyncHandle:894) Command async send retry: index = 0, retryTimes = 2, cmdSet = 0, cmdId = 0 
[758.992][linker]-[Warn]-[DjiCommand_SendAsyncHandle:894) Command async send retry: index = 0, retryTimes = 1, cmdSet = 0, cmdId = 0 
[759.993][linker]-[Error]-[DjiCommand_SendAsyncHandle:906) Command async send error 0 
[759.993][linker]-[Error]-[DjiCommand_SendSync:489) sendSync callback timeout 
[759.993][adapter]-[Error]-[DjiPayloadNegotiate_CheckNegotiatorReadyState:434) Set comm param sync error:0x000000E1. 
[759.993][adapter]-[Info]-[DjiPayloadNegotiate_Init:165) Waiting for negotiator to be ready... 
[761.994][linker]-[Warn]-[DjiCommand_SendAsyncHandle:894) Command async send retry: index = 0, retryTimes = 3, cmdSet = 0, cmdId = 0 
[762.995][linker]-[Warn]-[DjiCommand_SendAsyncHandle:894) Command async send retry: index = 0, retryTimes = 2, cmdSet = 0, cmdId = 0 
[763.996][linker]-[Warn]-[DjiCommand_SendAsyncHandle:894) Command async send retry: index = 0, retryTimes = 1, cmdSet = 0, cmdId = 0 
[764.997][linker]-[Error]-[DjiCommand_SendAsyncHandle:906) Command async send error 0 
[764.997][linker]-[Error]-[DjiCommand_SendSync:489) sendSync callback timeout 
[764.997][adapter]-[Error]-[DjiPayloadNegotiate_CheckNegotiatorReadyState:434) Set comm param sync error:0x000000E1. 
[764.997][adapter]-[Info]-[DjiPayloadNegotiate_Init:165) Waiting for negotiator to be ready... 
[766.998][linker]-[Warn]-[DjiCommand_SendAsyncHandle:894) Command async send retry: index = 0, retryTimes = 3, cmdSet = 0, cmdId = 0 
[767.999][linker]-[Warn]-[DjiCommand_SendAsyncHandle:894) Command async send retry: index = 0, retryTimes = 2, cmdSet = 0, cmdId = 0 
[769.000][linker]-[Warn]-[DjiCommand_SendAsyncHandle:894) Command async send retry: index = 0, retryTimes = 1, cmdSet = 0, cmdId = 0 
[770.001][linker]-[Error]-[DjiCommand_SendAsyncHandle:906) Command async send error 0 
[770.001][linker]-[Error]-[DjiCommand_SendSync:489) sendSync callback timeout 
[770.001][adapter]-[Error]-[DjiPayloadNegotiate_CheckNegotiatorReadyState:434) Set comm param sync error:0x000000E1.

dji-dev, 2024-07-05:
Agent comment from Leon in <a href='https://djisdksupport.zendesk.com/agent/tickets/111301'>Zendesk ticket #111301</a>:<br><br>尊敬的开发者，

您好，感谢您联系DJI 大疆创新。

您这个报错是负载协商的报错，PSDK主线程还没有拉起来，请问您的串口连接是直接连在板载串口上还是使用USB-TTL？

感谢您对大疆产品的支持！祝您一切顺利！

Best Regards,
DJI 大疆创新SDK技术支持
<br><p><em><small>°°°</small></em></p>

sthkklog, 2024-07-09:
串口连接是使用USB-TTL

dji-dev, 2024-07-09:
Agent comment from Leon in <a href='https://djisdksupport.zendesk.com/agent/tickets/111301'>Zendesk ticket #111301</a>:<br><br>尊敬的开发者，

您好，感谢您联系DJI 大疆创新。

您可以先参考下这篇文章，确保下硬件连接没有问题。
PSDK相关机型连接

感谢您对大疆产品的支持！祝您一切顺利！

Best Regards,
DJI 大疆创新SDK技术支持
<br><p><em><small>°°°</small></em></p>

sthkklog, 2024-07-09:
您好，文章链接不存在，看不了

sthkklog, 2024-07-09:
您好，文章链接不存在，看不了

dji-dev, 2024-07-10:
Agent comment from Leon in <a href='https://djisdksupport.zendesk.com/agent/tickets/111301'>Zendesk ticket #111301</a>:<br><br>尊敬的开发者，

您好，感谢您联系DJI 大疆创新。

不好意思，您看这篇。
PSDK各机型连接(M350/M300/M30/M30T/M3E/M3T/M3D/M3TD)

感谢您对大疆产品的支持！祝您一切顺利！

Best Regards,
DJI 大疆创新SDK技术支持
<br><p><em><small>°°°</small></em></p>

sthkklog, 2024-07-11:
您好，检测了硬件连接没问题。我们只是通过串口传输imu等数据时，psdk是问题的。但是通过网口传输图像数据后，psdk经常保存，要尝试很多次才能成功

sthkklog, 2024-07-11:
您好，检测了硬件连接没问题。我们只是通过串口传输imu等数据时，psdk是问题的。但是通过网口传输图像数据后，psdk经常报错，要尝试很多次才能成功

dji-dev, 2024-07-11:
Agent comment from Leon in <a href='https://djisdksupport.zendesk.com/agent/tickets/111301'>Zendesk ticket #111301</a>:<br><br>尊敬的开发者，

您好，感谢您联系DJI 大疆创新。

请问一下您NETWORK链路使用的是USB RNDIS还是USB网卡的连接方式？您在运行PSDK时，可以另起一个新的终端，ping一下飞机ip，看是不是有断连的现象，然后再排查是否为连接问题，飞机ip可以通过DjiPayloadCamera_GetVideoStreamRemoteAddress函数打印得知。

感谢您对大疆产品的支持！祝您一切顺利！

Best Regards,
DJI 大疆创新SDK技术支持
<br><p><em><small>°°°</small></em></p>

sthkklog, 2024-07-12:
NETWORK链路使用的是USB RNDIS,我们成功运行psdk后，ping飞机ip不会丢包；但psdk运行失败的情况下，ping飞机ip丢包很严重。还有就是需要运行psdk才能ping通飞机ip，不运行psdk的话ping不通psdk，这是正常的吗？

dji-dev, 2024-07-12:
Agent comment from Leon in <a href='https://djisdksupport.zendesk.com/agent/tickets/111301'>Zendesk ticket #111301</a>:<br><br>尊敬的开发者，

您好，感谢您联系DJI 大疆创新。

运行psdk以后才能ping通飞机，这个是符合预期的，如果psdk运行失败的时候，可以留意下网口ip是否掉线了，如果存在多个网卡，可以暂时先下掉，看看是不是网卡相互干扰导致的。

感谢您对大疆产品的支持！祝您一切顺利！

Best Regards,
DJI 大疆创新SDK技术支持
<br><p><em><small>°°°</small></em></p>

sthkklog, 2024-07-16:
您好，现发现板卡采用orin NX会出现上述的问题，采用性能较差的NX就没有这个问题，是什么原因呢？

dji-dev, 2024-07-16:
Agent comment from Leon in <a href='https://djisdksupport.zendesk.com/agent/tickets/111301'>Zendesk ticket #111301</a>:<br><br>尊敬的开发者，

您好，感谢您联系DJI 大疆创新。

很抱歉，目前我们还没有收到类似的反馈，这个建议还是先排查一下网络环境，您也可以尝试使用USB网卡来进行验证，看是否板载网卡兼容性导致。

感谢您对大疆产品的支持！祝您一切顺利！

Best Regards,
DJI 大疆创新SDK技术支持
<br><p><em><small>°°°</small></em></p>

javilinos, 2025-02-07:
Hello @dji-dev @sthkklog,

I am getting exactly the same diagnosis. I have the same environment: M30T, AGX Orin and PSDK v3.8.1. I have the same error logs as @sthkklog, but when I put the AGX Orin in 15W mode, then it just works perfectly. I do not correlate what could be happening here. 

By the way, the problem is not in the network configuration. When I used DJI_USE_ONLY_UART, I get the same error. 

Regards.