dji-sdk/Payload-SDK #184: DjiWaypointV2_Init() issue on Mavic 3E using PSDK 3.8

xenogit, 2024-07-12:
I am trying to do waypoint flying on the Mavic 3E using PSDK 3.8.

There is this Error message when calling DjiWaypointV2_Init().
**"Don't support this module on this aircraft series and mount position."**

I thought the PSDK supports waypoint flying?

Need advice as it is hard to look for relevant information.

dji-dev, 2024-07-12:
Agent comment from Leon in <a href='https://djisdksupport.zendesk.com/agent/tickets/111822'>Zendesk ticket #111822</a>:<br><br>Dear xenogit,

I hope this message finds you well. Thank you for reaching out to DJI Innovations.

Please be informed that the M3E model only supports Waypoint Mission V3 version and does not support V2 version. You may switch to the V3 version and try again.

Thank you for your support of DJI products! Wishing you all the best!

Best Regards,

DJI Innovations SDK Technical Support
<br><p><em><small>°°°</small></em></p>

xenogit, 2024-07-15:
Thank you for the reply. I will try it out.

However, from the documentation, I understand that Waypoint v3 uses KMZ files.

Understand that the KML file contents need to be in a certain format, followed by zipping the file and renaming the file to .kmz extension.

I got 2 follow up questions regarding Waypoint v3.
1) Is there a library function from PSDK that we can use to generate these KML/KMZ files?
2) Is there any specific zipping format required before we rename the zip files to .kmz extension?

dji-dev, 2024-07-15:
Agent comment from Leon in <a href='https://djisdksupport.zendesk.com/agent/tickets/111822'>Zendesk ticket #111822</a>:<br><br>Dear xenogit,

Greetings. Thank you for reaching out to DJI Innovations.

1. We regret to inform you that the requested files are not available. You can generate these files through the remote controller or write code to parse the content based on this guideline: https://developer.dji.com/doc/cloud-api-tutorial/en/api-reference/dji-wpml/waylines-wpml.html
2. There is no specific requirement for any compression format.

We appreciate your support for DJI products! Wishing you all the best!

Best Regards,

DJI Innovations SDK Technical Support
<br><p><em><small>°°°</small></em></p>