dji-sdk/Payload-SDK #186: PSDK control not possible w/o GPS & Visual Positioning

tinamumark, 2024-07-16:
Dear Community 

We observed the following behaviour. If we are not connected to GPS and turn off visual positioning using the remote control, we can no longer control the drone using the PSDK (Mavic 3E). 
However, requesting control authority and also DjiFlightController_ExecuteJoystickAction both return success. See below the joystick mode we are using. Also subscribing to DJI_FC_SUBSCRIPTION_TOPIC_CONTROL_DEVICE shows that PSDK should be in control. 
I’d also like to point out that we can fly without any issues as long as Visual Positioning is turned on (and working).
Do you have any idea why this happens or how we could overcome this issue?

**Background**: We are developing software for autonomous indoor flights in challenging environments. To support a new generation of drones, we decided to onboard the Mavic 3 Enterprise. While testing at a client’s side we observed that the flight mode might switch from N mode to Attitude mode when we fly with low ambient light present.  If we fly through an area where the drone switches to attitude mode, i.e. we do not manually change the mode, we lose control and can no longer control the drone until visual positioning works again. For more details on the issue you can also have a look here: https://sdk-forum.dji.net/hc/en-us/community/posts/34974338859929-SDK-control-in-Attitude-mode-not-possible

**Drone**: Mavic 3E
**PSDK version**: latest Github release
**Joystick mode setup**:
```
T_DjiFlightControllerJoystickMode joystickMode = {
       DJI_FLIGHT_CONTROLLER_HORIZONTAL_ANGLE_CONTROL_MODE,
       DJI_FLIGHT_CONTROLLER_VERTICAL_VELOCITY_CONTROL_MODE,
       DJI_FLIGHT_CONTROLLER_YAW_ANGLE_RATE_CONTROL_MODE,
       DJI_FLIGHT_CONTROLLER_HORIZONTAL_BODY_COORDINATE,
       DJI_FLIGHT_CONTROLLER_STABLE_CONTROL_MODE_ENABLE, // Also tried with DISABLE
   };
DjiFlightController_SetJoystickMode(joystickMode);
```

dji-dev, 2024-07-16:
Agent comment from Leon in <a href='https://djisdksupport.zendesk.com/agent/tickets/112055'>Zendesk ticket #112055</a>:<br><br>Dear tinamumark,

Greetings. We appreciate your contact with DJI - Da-Jiang Innovations.

We regret any inconvenience caused and have noticed that another developer has raised the same issue on the forum. We will investigate this matter. If the issue on the forum is the same as yours, we recommend continuing the discussion on the same ticket to avoid any confusion.

Thank you for your support of DJI products! We wish you all the best!

Best Regards,

DJI - Da-Jiang Innovations SDK Technical Support
<br><p><em><small>°°°</small></em></p>

joelreymont, 2024-08-02:
@tinamumark @tnaegeli Any updates on this?

Also, have you tried starting in ATTI mode with no visual positioning?