dji-sdk/Payload-SDK #187: 如果不使用USB-TTL，直接用杜邦线连UART，需要如何配置

github4529, 2024-07-17:
在hal_uart.h文件中有这个配置
```
#define LINUX_UART_DEV1    "/dev/ttyUSB0"
#define LINUX_UART_DEV2    "/dev/ttyACM0"
```
如果不使用USB-TTL，直接用杜邦线连UART，需要如何配置
如果UART对应是/dev/ttyS0，将LINUX_UART_DEV1配置为/dev/ttyS0就行吗
LINUX_UART_DEV2需要如何配置呢
硬件：
rk3588，M350

dji-dev, 2024-07-17:
Agent comment from Leon in <a href='https://djisdksupport.zendesk.com/agent/tickets/112160'>Zendesk ticket #112160</a>:<br><br>尊敬的开发者，

您好，感谢您联系DJI 大疆创新。

使用杜邦线连接也是需要查看ls /dev/tty*的端口是否有新增，通过这个来配置DEV1，M350机型不需要配置DEV2。
PSDK各机型连接(M350/M300/M30/M30T/M3E/M3T/M3D/M3TD)

感谢您对大疆产品的支持！祝您一切顺利！

Best Regards,
DJI 大疆创新SDK技术支持
<br><p><em><small>°°°</small></em></p>

github4529, 2024-07-17:
启动飞机后，查看ls /dev/tty*的端口并没有新增，板子配置的uart0是/dev/ttyS0，将h文件的dev1配置成/dev/ttyS0后，启动程序显示 Try identify UART0 connection failed，直接使用杜邦线连接是否需要其他配置，需要rk3588烧录的时候配置将UART0开放之类的

github4529, 2024-07-17:
使用USB-TTL，波特率会被PSDK设置为921600
使用杜邦线直连串口和E开发套件，波特率一直在变化
460800
230400
921600
1000000
```
stty -F /dev/ttyS0
speed xxxx baud; line = 0;
min = 0; time = 0;
-brkint -icrnl -imaxbel
-opost
-isig -icanon -echo -echoe
```

github4529, 2024-07-17:
如果不使用USB-TTL，直接用杜邦线直连串口-EPort
GND-GND
RX-TX
TX-RX
还需要别的连线吗

github4529, 2024-07-17:
EPort开发者套件的串口是TTL电平吗

dji-dev, 2024-07-17:
Agent comment from Leon in <a href='https://djisdksupport.zendesk.com/agent/tickets/112160'>Zendesk ticket #112160</a>:<br><br>尊敬的开发者，

您好，感谢您联系DJI 大疆创新。

E-Port Uart需遵循 3.3V TTL 协议。接入RX TX GND即可。关于使用板载串口通信，您可能要查阅下相关的产品手册看怎么使能板载串口，并且需与/dev/tty* 的端口对应上。例如GPIO2 GPIO3 对应端口/dev/ttyAMA1这样的对应关系要确定。

感谢您对大疆产品的支持！祝您一切顺利！

Best Regards,
DJI 大疆创新SDK技术支持<br><a href='https://djisdksupport.zendesk.com/attachments/token/2tcSrN3ObAVsu4HjBUNJZCgIL' target='zd_attach'>image.png</a>
<br><p><em><small>°°°</small></em></p>