dji-sdk/Payload-SDK #190: Jetson ORIN NX在M350上，通过E-PORT无法查看相机视频流

Flying-fat-sloth, 2024-08-01:
你好，我们使用Jetson ORIN NX通过E-port连接了M350。
硬件连接方案为：
（1）E-port的串口通过USB-串口模块连接Jetson ORIN NX的USB接口
（2）Jetson ORIN NX和E-port的type-C接口使用双type-C头的线连接（Jetson ORIN NX的type-C口支持USB Device模式）
参考资料：
（1）官方指导手册
（2） [树莓派4B配置USB device RNDIS 和 BULK](https://sdk-forum.dji.net/hc/zh-cn/articles/10232604141465-%E6%A0%91%E8%8E%93%E6%B4%BE4B%E9%85%8D%E7%BD%AEUSB-device-RNDIS-%E5%92%8C-BULK) 
（3）[video stream not working in M3TD #180](https://github.com/dji-sdk/Payload-SDK/issues/180) 
目前的状态及问题：
（1）按照官方指导配置了所有信息，基于UART的功能都可以验证；
（2）配置了USB bulk，上述参考资料（2）最后给出的验证USB Bulk链路的程序，可以在我们的Jetson ORIN NX上自环收发，如下图
[image]
（3）基于PSDK中C++工程的nvidia_jetson示例程序， [c] Camera stream view sample - display the camera video stream，无法收到图像数据，现象与上述参考资料（3）描述相同，如下图：
[image]
通过在示例程序中添加调试输出，发现示例程序在我们选择 [c] Camera stream view sample - display the camera video stream后，程序不仅进行端口信息读取和初始化，还调用了hal_usb_bulk.c文件中的T_DjiReturnCode HalUsbBulk_ReadData(T_DjiUsbBulkHandle usbBulkHandle, uint8_t *buf, uint32_t len）函数，同时程序在 *realLen = read(((T_HalUsbBulkObj *) usbBulkHandle)->ep2, buf, len);
出错或停止。但是，这里的函数与BULK验证程序中几乎完全一致，在那个程序中正常，在这里却不好使。
（4）我们进一步配置编译了C++工程下的maniflod2示例程序，得到了几乎相同的结果。
请问我还能做些什么？或者还有什么环节没有按照要求实现。
谢谢！

dji-dev, 2024-08-01:
Agent comment from Leon in <a href='https://djisdksupport.zendesk.com/agent/tickets/113494'>Zendesk ticket #113494</a>:<br><br>尊敬的开发者，

您好，感谢您联系DJI 大疆创新。

很抱歉为您带来不好的体验，请问一下您是否有检查过hal_usb_bulk.h中的配置与您的测试节点是否一致？
您可以使用DjiTest_LiveviewRunSample函数进一步验证，在DjiTest_FpvCameraStreamCallback或DjiTest_PayloadCameraStreamCallback中加入打印，看调用时是否会进入回调函数，如果没有进入回调函数，表明bulk链路还有问题。

感谢您对大疆产品的支持！祝您一切顺利！

Best Regards,
DJI 大疆创新SDK技术支持
<br><p><em><small>°°°</small></em></p>

Flying-fat-sloth, 2024-08-01:
您好：
     来信已收到，我会尽快处理。
                                          ——陈云浩

Flying-fat-sloth, 2024-08-02:
您好，我仔细检查了hal_usb_bulk.h中的配置，与测试节点完全一致，且通过测试程序对bulk1和bulk2两个通道均进行了测试。对程序中多个函数进行了输出验证，发现程序成功注册了回调函数并创建了新的线程，但未调用（进入）回调函数，仍无法看到回传画面。
USB数据接收似乎被阻塞在hal_usb_bulk.c的HalUsbBulk_ReadData函数中，没有读到数据。
请问还有可能存在哪些问题？
如果bulk链路还有问题， 应当如何排查？
谢谢！

dji-dev, 2024-08-05:
Agent comment from Leon in <a href='https://djisdksupport.zendesk.com/agent/tickets/113494'>Zendesk ticket #113494</a>:<br><br>尊敬的开发者，

您好，感谢您联系DJI 大疆创新。

您还可以检查下VID PID是否为0955 7020，这个有可能会导致异常，您的节点如果测试收发没问题，不排除是飞机硬件异常，您也可以寄回售后团队处检测：
https://repair.dji.com/cn/repair/index
另外请问下您是否有检查过妙算工程中的dji_sdk_config.h文件？看看配置是否有切换过来。您是否还有配置RNDIS？也可以用RNDIS来尝试看是否正常。

感谢您对大疆产品的支持！祝您一切顺利！

Best Regards,
DJI 大疆创新SDK技术支持
<br><p><em><small>°°°</small></em></p>

Flying-fat-sloth, 2024-08-06:
感谢帮助，我们已经可以顺利运行demo，看到回传视频了。
我们出现的问题在于必须使用Type-C的OTG线插在E-Port上，再通过USB-A===Type-C的线连接到Jetson ORIN NX的Type-C端口上。
使用双头Type-C直接连接E-Port和Jetson无法接收到数据。
以上，希望可以帮助到其他人。

dji-dev, 2024-08-06:
Agent comment from Leon in <a href='https://djisdksupport.zendesk.com/agent/tickets/113494'>Zendesk ticket #113494</a>:<br><br>尊敬的开发者，

您好，感谢您联系DJI 大疆创新。

感谢您的反馈，我们会将您的问题记录，以便能够给更多开发者参考。

感谢您对大疆产品的支持！祝您一切顺利！

Best Regards,
DJI 大疆创新SDK技术支持
<br><p><em><small>°°°</small></em></p>

MrLiuXiangzhen, 2024-08-13:
我也遇到了一样的问题，楼主最终是只换了线解决了吗，我用的老版本的NX没有TypeC接口，尝试用OTG中转E-Port也没有解决，PSDK的云台和相机其它功能都正常的

mingren11, 2024-08-30:
我也遇到了同样的问题

blackjacket996, 2024-12-03:
请问Orin NX进行USB HULK链路的时候是按照官方给的E-port nano上手指南步骤的么

19880020635, 2025-11-15:
你好，我同样遇到无法收到图像数据，你的配置步骤，能告知我吗

sadassdadadas, 2025-12-16:
# @Flying-fat-sloth 请问具体的线缆连接方式能不能详细描述一下，我也遇到这样的问题，谢谢！

GrooveWJH, 2025-12-18:
> # [@Flying-fat-sloth](https://github.com/Flying-fat-sloth) 请问具体的线缆连接方式能不能详细描述一下，我也遇到这样的问题，谢谢！

简而言之就是使用 C转A母口 （也就是OTG线）连接到eport上，然后另拿一根A to C 的线连接到伴随计算机上。因为psdk的通讯使用了usb bulk 和 RNDIS，需要让伴随计算机属于从设备，Eport属于主设备（文档里讲的是M300 RTK 除外，应当让eport进入device模式）

但为何c to c 就无法工作，这背后的机制是：
> **把“难搞的 Type-C 协商”替换成“传统 USB-A 的确定性”**。

1. **USB-A 这一段天然固定主从关系**  

    USB-A 没有 CC，也没有正反插，谁是 Host 基本是物理形态决定的。  

    所以 E-Port（Host）→ USB-A 后，链路行为更像“老 USB2”，稳定得多。
2. **转接/OTG + A-to-C 线会让 Jetson 更确定地落到 Device/UFP**  

    很多 “A-to-C（主机A → 设备C）” 的链路，会让 C 口那边更明确地进入“作为外设被枚举”的路径（至少在工程实现上更容易触发 Jetson 的 gadget/device 逻辑），避免它再去当 Host 抢角色。