dji-sdk/Payload-SDK #191: Compatibility with Canon camera

GregCornis, 2024-08-06:
Hi,
We're wondering if we can use the PSDK to connect a Canon (R6) camera to a M300 RTK, via a microcontroller. Does that seem doable ? Has anyone done a similar thing already ?
We need to have the live stream, be able to take picture and configure some settings, like exposure and focus.
Thanks !

GregCornis, 2024-08-08:
Ok thanks, I'll give it a try. Are you interested in the result so that you can share it with others ?

dji-dev, 2024-08-08:
Agent comment from Leon in <a href='https://djisdksupport.zendesk.com/agent/tickets/113849'>Zendesk ticket #113849</a>:<br><br>Dear Developer,

Greetings, and thank you for contacting DJI.

We look forward to your results, and if your integration solution is successful, with your permission, we would like to share it with more developers for their reference or as a selection recommendation.

Thank you for your support of DJI products! We wish you all the best!

Best Regards,

DJI SDK Technical Support
<br><p><em><small>°°°</small></em></p>