dji-sdk/Payload-SDK #198: MDK编译报错，请问是我环境配置的问题吗，为什么这么多变量显示未定义？

Jtongleong, 2024-08-15:
..\..\Drivers\STM32F4xx_HAL_Driver\Src\stm32f4xx_hal.c(526): error:  #20: identifier "SYSCFG_CMPCR_CMP_PD_Pos" is undefined
    *(__IO uint32_t *)CMPCR_CMP_PD_BB = (uint32_t)ENABLE;
..\..\Drivers\STM32F4xx_HAL_Driver\Src\stm32f4xx_hal.c(537): error:  #20: identifier "SYSCFG_CMPCR_CMP_PD_Pos" is undefined
    *(__IO uint32_t *)CMPCR_CMP_PD_BB = (uint32_t)DISABLE;
..\..\Drivers\STM32F4xx_HAL_Driver\Src\stm32f4xx_hal.c(546): error:  #20: identifier "UID_BASE" is undefined
    return (READ_REG(*((uint32_t *)UID_BASE)));
..\..\Drivers\STM32F4xx_HAL_Driver\Src\stm32f4xx_hal.c(555): error:  #20: identifier "UID_BASE" is undefined
    return (READ_REG(*((uint32_t *)(UID_BASE + 4U))));
..\..\Drivers\STM32F4xx_HAL_Driver\Src\stm32f4xx_hal.c(564): error:  #20: identifier "UID_BASE" is undefined
    return (READ_REG(*((uint32_t *)(UID_BASE + 8U))));
..\..\Drivers\STM32F4xx_HAL_Driver\Src\stm32f4xx_hal.c: 0 warnings, 5 errors

..\..\Drivers\STM32F4xx_HAL_Driver\Src\stm32f4xx_hal_flash_ex.c(1011): error:  #20: identifier "FLASH_CR_SNB_Pos" is undefined
    FLASH->CR |= FLASH_CR_SER | (Sector << FLASH_CR_SNB_Pos);
..\..\Drivers\STM32F4xx_HAL_Driver\Src\stm32f4xx_hal_flash_ex.c: 0 warnings, 1 error
compiling stm32f4xx_hal_exti.c...
..\..\Drivers\STM32F4xx_HAL_Driver\Src\stm32f4xx_hal_exti.c(201): error:  #20: identifier "SYSCFG_EXTICR1_EXTI1_Pos" is undefined
        regval &= ~(SYSCFG_EXTICR1_EXTI0 << (SYSCFG_EXTICR1_EXTI1_Pos * (linepos & 0x03u)));
..\..\Drivers\STM32F4xx_HAL_Driver\Src\stm32f4xx_hal_exti.c(305): error:  #20: identifier "SYSCFG_EXTICR1_EXTI1_Pos" is undefined
        pExtiConfig->GPIOSel = ((regval << (SYSCFG_EXTICR1_EXTI1_Pos * (3uL - (linepos & 0x03u)))) >> 24);
..\..\Drivers\STM32F4xx_HAL_Driver\Src\stm32f4xx_hal_exti.c(364): error:  #20: identifier "SYSCFG_EXTICR1_EXTI1_Pos" is undefined
        regval &= ~(SYSCFG_EXTICR1_EXTI0 << (SYSCFG_EXTICR1_EXTI1_Pos * (linepos & 0x03u)));
..\..\Drivers\STM32F4xx_HAL_Driver\Src\stm32f4xx_hal_exti.c: 0 warnings, 3 errors

..\..\Drivers\STM32F4xx_HAL_Driver\Src\stm32f4xx_hal_pwr_ex.c(193): error:  #20: identifier "PWR_CR_FPDS_Pos" is undefined
    *(__IO uint32_t *) CR_FPDS_BB = (uint32_t)ENABLE;
..\..\Drivers\STM32F4xx_HAL_Driver\Src\stm32f4xx_hal_pwr_ex.c(202): error:  #20: identifier "PWR_CR_FPDS_Pos" is undefined
    *(__IO uint32_t *) CR_FPDS_BB = (uint32_t)DISABLE;
..\..\Drivers\STM32F4xx_HAL_Driver\Src\stm32f4xx_hal_pwr_ex.c: 0 warnings, 4 errors
compiling stm32f4xx_hal_pwr.c...
..\..\Drivers\STM32F4xx_HAL_Driver\Src\stm32f4xx_hal_pwr.c(113): error:  #20: identifier "PWR_CR_DBP_Pos" is undefined
    *(__IO uint32_t *) CR_DBP_BB = (uint32_t)ENABLE;
..\..\Drivers\STM32F4xx_HAL_Driver\Src\stm32f4xx_hal_pwr.c(133): error:  #20: identifier "PWR_CR_DBP_Pos" is undefined
    *(__IO uint32_t *) CR_DBP_BB = (uint32_t)DISABLE;
..\..\Drivers\STM32F4xx_HAL_Driver\Src\stm32f4xx_hal_pwr.c(313): error:  #20: identifier "PWR_CR_PVDE_Pos" is undefined
    *(__IO uint32_t *) CR_PVDE_BB = (uint32_t)ENABLE;
..\..\Drivers\STM32F4xx_HAL_Driver\Src\stm32f4xx_hal_pwr.c(322): error:  #20: identifier "PWR_CR_PVDE_Pos" is undefined
    *(__IO uint32_t *) CR_PVDE_BB = (uint32_t)DISABLE;
..\..\Drivers\STM32F4xx_HAL_Driver\Src\stm32f4xx_hal_pwr.c: 0 warnings, 4 errors

..\..\Drivers\STM32F4xx_HAL_Driver\Src\stm32f4xx_hal_rcc.c(303): error:  #20: identifier "RCC_CR_HSITRIM_Pos" is undefined
          __HAL_RCC_HSI_CALIBRATIONVALUE_ADJUST(RCC_OscInitStruct->HSICalibrationValue);
..\..\Drivers\STM32F4xx_HAL_Driver\Src\stm32f4xx_hal_rcc.c(327): error:  #20: identifier "RCC_CR_HSITRIM_Pos" is undefined
          __HAL_RCC_HSI_CALIBRATIONVALUE_ADJUST(RCC_OscInitStruct->HSICalibrationValue);
..\..\Drivers\STM32F4xx_HAL_Driver\Src\stm32f4xx_hal_rcc.c(494): error:  #20: identifier "RCC_PLLCFGR_PLLN_Pos" is undefined
          WRITE_REG(RCC->PLLCFGR, (RCC_OscInitStruct->PLL.PLLSource                                            | \
..\..\Drivers\STM32F4xx_HAL_Driver\Src\stm32f4xx_hal_rcc.c(494): error:  #20: identifier "RCC_PLLCFGR_PLLP_Pos" is undefined
          WRITE_REG(RCC->PLLCFGR, (RCC_OscInitStruct->PLL.PLLSource                                            | \
..\..\Drivers\STM32F4xx_HAL_Driver\Src\stm32f4xx_hal_rcc.c(494): error:  #20: identifier "RCC_PLLCFGR_PLLQ_Pos" is undefined
          WRITE_REG(RCC->PLLCFGR, (RCC_OscInitStruct->PLL.PLLSource                                            | \
..\..\Drivers\STM32F4xx_HAL_Driver\Src\stm32f4xx_hal_rcc.c(554): error:  #20: identifier "RCC_PLLCFGR_PLLM_Pos" is undefined
              (READ_BIT(pll_config, RCC_PLLCFGR_PLLM) != (RCC_OscInitStruct->PLL.PLLM) << RCC_PLLCFGR_PLLM_Pos) ||
..\..\Drivers\STM32F4xx_HAL_Driver\Src\stm32f4xx_hal_rcc.c(555): error:  #20: identifier "RCC_PLLCFGR_PLLN_Pos" is undefined
              (READ_BIT(pll_config, RCC_PLLCFGR_PLLN) != (RCC_OscInitStruct->PLL.PLLN) << RCC_PLLCFGR_PLLN_Pos) ||
..\..\Drivers\STM32F4xx_HAL_Driver\Src\stm32f4xx_hal_rcc.c(556): error:  #20: identifier "RCC_PLLCFGR_PLLP_Pos" is undefined
              (READ_BIT(pll_config, RCC_PLLCFGR_PLLP) != (((RCC_OscInitStruct->PLL.PLLP >> 1U) - 1U)) << RCC_PLLCFGR_PLLP_Pos) ||
..\..\Drivers\STM32F4xx_HAL_Driver\Src\stm32f4xx_hal_rcc.c(557): error:  #20: identifier "RCC_PLLCFGR_PLLQ_Pos" is undefined
              (READ_BIT(pll_config, RCC_PLLCFGR_PLLQ) != (RCC_OscInitStruct->PLL.PLLQ << RCC_PLLCFGR_PLLQ_Pos)))
..\..\Drivers\STM32F4xx_HAL_Driver\Src\stm32f4xx_hal_rcc.c(683): error:  #20: identifier "RCC_CFGR_SWS_Pos" is undefined
      while (__HAL_RCC_GET_SYSCLK_SOURCE() != (RCC_ClkInitStruct->SYSCLKSource << RCC_CFGR_SWS_Pos))
..\..\Drivers\STM32F4xx_HAL_Driver\Src\stm32f4xx_hal_rcc.c(721): error:  #20: identifier "AHBPrescTable" is undefined
    SystemCoreClock = HAL_RCC_GetSysClockFreq() >> AHBPrescTable[(RCC->CFGR & RCC_CFGR_HPRE)>> RCC_CFGR_HPRE_Pos];
..\..\Drivers\STM32F4xx_HAL_Driver\Src\stm32f4xx_hal_rcc.c(721): error:  #20: identifier "RCC_CFGR_HPRE_Pos" is undefined
    SystemCoreClock = HAL_RCC_GetSysClockFreq() >> AHBPrescTable[(RCC->CFGR & RCC_CFGR_HPRE)>> RCC_CFGR_HPRE_Pos];
..\..\Drivers\STM32F4xx_HAL_Driver\Src\stm32f4xx_hal_rcc.c(913): error:  #20: identifier "RCC_PLLCFGR_PLLN_Pos" is undefined
          pllvco = (uint32_t) ((((uint64_t) HSE_VALUE * ((uint64_t) ((RCC->PLLCFGR & RCC_PLLCFGR_PLLN) >> RCC_PLLCFGR_PLLN_Pos)))) / (uint64_t)pllm);
..\..\Drivers\STM32F4xx_HAL_Driver\Src\stm32f4xx_hal_rcc.c(918): error:  #20: identifier "RCC_PLLCFGR_PLLN_Pos" is undefined
          pllvco = (uint32_t) ((((uint64_t) HSI_VALUE * ((uint64_t) ((RCC->PLLCFGR & RCC_PLLCFGR_PLLN) >> RCC_PLLCFGR_PLLN_Pos)))) / (uint64_t)pllm);
..\..\Drivers\STM32F4xx_HAL_Driver\Src\stm32f4xx_hal_rcc.c(920): error:  #20: identifier "RCC_PLLCFGR_PLLP_Pos" is undefined
        pllp = ((((RCC->PLLCFGR & RCC_PLLCFGR_PLLP) >> RCC_PLLCFGR_PLLP_Pos) + 1U) *2U);
[... 2 lines cut ...]
..\..\Drivers\STM32F4xx_HAL_Driver\Src\stm32f4xx_hal_rcc.c(957): error:  #20: identifier "RCC_CFGR_PPRE1_Pos" is undefined
    return (HAL_RCC_GetHCLKFreq() >> APBPrescTable[(RCC->CFGR & RCC_CFGR_PPRE1)>> RCC_CFGR_PPRE1_Pos]);
..\..\Drivers\STM32F4xx_HAL_Driver\Src\stm32f4xx_hal_rcc.c(969): error:  #20: identifier "APBPrescTable" is undefined
    return (HAL_RCC_GetHCLKFreq()>> APBPrescTable[(RCC->CFGR & RCC_CFGR_PPRE2)>> RCC_CFGR_PPRE2_Pos]);
..\..\Drivers\STM32F4xx_HAL_Driver\Src\stm32f4xx_hal_rcc.c(969): error:  #20: identifier "RCC_CFGR_PPRE2_Pos" is undefined
    return (HAL_RCC_GetHCLKFreq()>> APBPrescTable[(RCC->CFGR & RCC_CFGR_PPRE2)>> RCC_CFGR_PPRE2_Pos]);
..\..\Drivers\STM32F4xx_HAL_Driver\Src\stm32f4xx_hal_rcc.c(1008): error:  #20: identifier "RCC_CR_HSITRIM_Pos" is undefined
    RCC_OscInitStruct->HSICalibrationValue = (uint32_t)((RCC->CR &RCC_CR_HSITRIM) >> RCC_CR_HSITRIM_Pos);
..\..\Drivers\STM32F4xx_HAL_Driver\Src\stm32f4xx_hal_rcc.c(1045): error:  #20: identifier "RCC_PLLCFGR_PLLN_Pos" is undefined
    RCC_OscInitStruct->PLL.PLLN = (uint32_t)((RCC->PLLCFGR & RCC_PLLCFGR_PLLN) >> RCC_PLLCFGR_PLLN_Pos);
..\..\Drivers\STM32F4xx_HAL_Driver\Src\stm32f4xx_hal_rcc.c(1046): error:  #20: identifier "RCC_PLLCFGR_PLLP_Pos" is undefined
    RCC_OscInitStruct->PLL.PLLP = (uint32_t)((((RCC->PLLCFGR & RCC_PLLCFGR_PLLP) + RCC_PLLCFGR_PLLP_0) << 1U) >> RCC_PLLCFGR_PLLP_Pos);
..\..\Drivers\STM32F4xx_HAL_Driver\Src\stm32f4xx_hal_rcc.c(1047): error:  #20: identifier "RCC_PLLCFGR_PLLQ_Pos" is undefined
    RCC_OscInitStruct->PLL.PLLQ = (uint32_t)((RCC->PLLCFGR & RCC_PLLCFGR_PLLQ) >> RCC_PLLCFGR_PLLQ_Pos);
[... cut ...]
..\..\Drivers\STM32F4xx_HAL_Driver\Src\stm32f4xx_hal_rcc_ex.c(2651): error:  #20: identifier "RCC_PLLI2SCFGR_PLLI2SN_Pos" is undefined
    PeriphClkInit->PLLI2S.PLLI2SN = (uint32_t)((RCC->PLLI2SCFGR & RCC_PLLI2SCFGR_PLLI2SN) >> RCC_PLLI2SCFGR_PLLI2SN_Pos);
..\..\Drivers\STM32F4xx_HAL_Driver\Src\stm32f4xx_hal_rcc_ex.c(2652): error:  #20: identifier "RCC_PLLI2SCFGR_PLLI2SR_Pos" is undefined
    PeriphClkInit->PLLI2S.PLLI2SR = (uint32_t)((RCC->PLLI2SCFGR & RCC_PLLI2SCFGR_PLLI2SR) >> RCC_PLLI2SCFGR_PLLI2SR_Pos);
..\..\Drivers\STM32F4xx_HAL_Driver\Src\stm32f4xx_hal_rcc_ex.c: 0 warnings, 4 errors

dji-dev, 2024-08-15:
Agent comment from Leon in <a href='https://djisdksupport.zendesk.com/agent/tickets/114666'>Zendesk ticket #114666</a>:<br><br>您好，这个是您环境配置的问题，....\Drivers\STM32F4xx_HAL_Driver\Src\stm32f4xx_hal.c(537): error: #20: identifier "SYSCFG_CMPCR_CMP_PD_Pos" is undefined
这个目录是STM32的代码，不是SDK代码。
<br><p><em><small>°°°</small></em></p>

zhanqk, 2025-06-19:
缺少包含正确的头文件。..\..\drivers\CMSIS\Device\ST\STM32F4xx\Include