dji-sdk/Payload-SDK #201: Values of ESC current and voltage

nobodyinperson, 2024-08-28:
I wonder if the specified ESC current and voltage units are correct.

From `dji_fc_subscription.h`:

```c++
  /*!
   * @brief struct for status of each individual esc
   */
  typedef struct ESCStatusIndividual
  {
    int16_t current;              /*!< ESC current, unit: mA */
    int16_t speed;                /*!< ESC speed, unit: rpm */
    uint16_t voltage;             /*!< Input power from battery to ESC, unit: mV */
    int16_t temperature;          /*!< ESC temperature, unit: degree C */
    uint16_t stall : 1;           /*!< Motor is stall */
    uint16_t empty : 1;           /*!< Motor has no load */
    uint16_t unbalanced : 1;      /*!< Motor speed is unbalanced */
    uint16_t escDisconnected : 1; /*!< ESC is disconnected */
    uint16_t temperatureHigh : 1; /*!< Temperature is high */
    uint16_t reserved : 11;
  } ESCStatusIndividual;
```

However, with our Mavic 3E, I get (`printf("%d",esc_data.esc[0].voltage)` etc.) values like `1634` for the ESC voltages (multiplied by 10 they get very close to the battery voltage, so I guess a factor of 10 is missing) and for the ESC currents, the values are very low, `400` to `900` or so. This doesn't check out with the battery current which is around `10000` mA (10 A). Here, a simple ×10 ×4 motors of the current yields too much, so something else is weird here.

Any ideas?

dji-dev, 2024-08-29:
Agent comment from Leon in <a href='https://djisdksupport.zendesk.com/agent/tickets/115811'>Zendesk ticket #115811</a>:<br><br>Dear Developer,

Thank you for reaching out to DJI.

We sincerely apologize for any inconvenience this may have caused. May I kindly inquire which version of the PSDK you are currently using? This may potentially be a bug within the SDK.

We appreciate your support for DJI products. Wishing you all the best!

Best Regards,

DJI SDK Technical Support
<br><p><em><small>°°°</small></em></p>

nobodyinperson, 2024-08-29:
Hi,

This is copy-pasted from the `dji_version.h` I use:

```c
/* Exported constants --------------------------------------------------------*/
#define DJI_VERSION_MAJOR 3 /*!< DJI SDK major version num, when have incompatible API changes. Range from 0 to 99. */
#define DJI_VERSION_MINOR                                                                                              \
  9 /*!< DJI SDK minor version num, when add functionality in a backwards compatible manner changes. Range from 0      \
       to 99. */
#define DJI_VERSION_MODIFY                                                                                             \
  0 /*!< DJI SDK modify version num, when have backwards compatible bug fixes changes. Range from 0 to 99. */
#define DJI_VERSION_BETA                                                                                               \
  0 /*!< DJI SDK version beta info, release version will be 0, when beta version release changes. Range from 0 to 255. \
     */
#define DJI_VERSION_BUILD                                                                                              \
  2044 /*!< DJI SDK version build info, when jenkins trigger build changes. Range from 0 to 65535. */
```

When I run the program, it outputs

```log
[0.019][core]-[Info]-[DjiCore_Init:106) Payload SDK Version : V3.9.0-beta.0-build.2044
```

Thank you for looking into this!

dji-dev, 2024-08-30:
Agent comment from Leon in <a href='https://djisdksupport.zendesk.com/agent/tickets/115811'>Zendesk ticket #115811</a>:<br><br>Dear nobodyinperson,

Thank you for reaching out to DJI.

We appreciate the additional information you have provided. Would you be so kind as to attach the flight control logs and the PSDK debug logs? We would like to confirm this with our development team.

Thank you for your support of DJI products. Wishing you all the best!

Best Regards,

DJI SDK Technical Support
<br><p><em><small>°°°</small></em></p>

nobodyinperson, 2024-08-30:
Please find in attached archive the `.dat` log from a recent flight and our custom CSV logfile which includes data from the PSDK port. We don't log the PSDK debug messages, sorry.

[flight.zip](https://github.com/user-attachments/files/16814793/flight.zip)