dji-sdk/Payload-SDK #203: 通过Payload SDK Board连接树莓派的网口

my-run, 2024-09-05:
[image]
正在设计一款M350RTK（已到货）用的成像负载，挂载X-port（已到货）云台连接相机。
有些疑问还请回复：
1. 通过X-port连接Payload SDK Board，成像负载连接树莓派，可否通过树莓派的网口与Payload SDK Board的网口连接实现推流？
2. hal_network.h中的USB_NET_ADAPTER_VID和USB_NET_ADAPTER_PID和树莓派的eth0有关系吗？
Thanks！

dji-dev, 2024-09-06:
Agent comment from Leon in <a href='https://djisdksupport.zendesk.com/agent/tickets/116493'>Zendesk ticket #116493</a>:<br><br>你好，
1. 通过X-port连接Payload SDK Board，成像负载连接树莓派，可否通过树莓派的网口与Payload SDK Board的网口连接实现推流？ 2. hal_network.h中的USB_NET_ADAPTER_VID和USB_NET_ADAPTER_PID和树莓派的eth0有关系吗？
1、可以通过网口与PSDK的开发套件网口连接来实现推流。
2、如果使用的是网口的端口（eth x），不需要配置VID PID（配不配置没有影响），这个是针对于运行在E-Port端口的PSDK程序来说的，此处的VID PID配置的是USB网卡的信息。
<br><p><em><small>°°°</small></em></p>

my-run, 2024-09-06:
谢谢回复。

已经编译了样例程序，并连接到了无人机。
请问老是提示如图错误，是什么问题？
[image]

dji-dev, 2024-09-06:
Agent comment from Leon in <a href='https://djisdksupport.zendesk.com/agent/tickets/116493'>Zendesk ticket #116493</a>:<br><br>您好，C++sample默认是运行在拓展口的，您的PSDK属于是Payload Port云台口，您可以在代码的此处把限制去掉。<br><a href='https://djisdksupport.zendesk.com/attachments/token/Oyz6YlHSYCs6hTf8Mi0D6pCK1' target='zd_attach'>image.png</a>
<br><p><em><small>°°°</small></em></p>

closed 2024-09-28 by my-run