dji-sdk/Payload-SDK #204: 3.9.1版本占用ram太大

dzx815, 2024-09-12:
芯片 stm32F401RET6

3.9.1版本，编译报这个错误
C:/ST/STM32CubeIDE_1.14.1/STM32CubeIDE/plugins/com.st.stm32cube.ide.mcu.externaltools.gnu-tools-for-stm32.12.3.rel1.win32_1.0.200.202406191623/tools/bin/../lib/gcc/arm-none-eabi/12.3.1/../../../../arm-none-eabi/bin/ld.exe: ap30_g2_psdk.elf section `.bss' will not fit in region `RAM'
C:/ST/STM32CubeIDE_1.14.1/STM32CubeIDE/plugins/com.st.stm32cube.ide.mcu.externaltools.gnu-tools-for-stm32.12.3.rel1.win32_1.0.200.202406191623/tools/bin/../lib/gcc/arm-none-eabi/12.3.1/../../../../arm-none-eabi/bin/ld.exe: region `RAM' overflowed by 3096 bytes
collect2.exe: error: ld returned 1 exit status

3.8.1版本是可以正常编译执行的，而且ram还剩余3kb多
[image]

而且我检测了一下内存使用量，比如自定义控件task，实际用到了400 word左右，但是你们默认分配了2048 word，至于静态lib库中具体分配的多少，我这边确实看不到。

dzx815, 2024-09-12:
3.9.1版本如下图所示，为什么定义了两个一样的bian'l变量，占用内存这么大
[image]

对比3.8.1版本
[image]

dji-dev, 2024-09-13:
Agent comment from Leon in <a href='https://djisdksupport.zendesk.com/agent/tickets/117138'>Zendesk ticket #117138</a>:<br><br>尊敬的开发者，

您好，感谢您联系DJI 大疆创新。

很抱歉为您带来不便，这个是内部新增的功能，3.9.1开始新增的。

感谢您对大疆产品的支持！祝您一切顺利！

Best Regards,
DJI 大疆创新SDK技术支持
<br><p><em><small>°°°</small></em></p>

closed 2024-09-18 by dzx815