dji-sdk/Payload-SDK #207: PSDK Controlling the YAW independent of mission.

Ellislee1, 2024-09-26:
We want to create a mission in the Pilot 2 app then via the PSDK we want to manually adjust the yaw (either by giving a heading directly or adjusting the yaw rate). We want the mission to be unaffected and continue normally just with the yaw being adjusted.

Looking at the demo/example code it appears there is a way to provide offsets:
```C
"--> Step 7: Move with north:3(m/s), east:0(m/s), up:0(m/s), yaw:0(deg/s) from current point for 2.5s!");
DjiTest_FlightControlVelocityAndYawRateCtrl((T_DjiTestFlightControlVector3f) {3, 0, 0}, 0, 2500);
``` 
[Line 535: test_flight_controls.c](https://github.com/dji-sdk/Payload-SDK/blob/159708e6ef8920f41dce11ce666a3446e91481f5/samples/sample_c/module_sample/flight_control/test_flight_control.c#L535)

Which calls:
```C
void DjiTest_FlightControlVelocityAndYawRateCtrl(const T_DjiTestFlightControlVector3f offsetDesired, float yawRate, 
        uint32_t timeMs)
```
[Line 1316: test_flight_controls.c](https://github.com/dji-sdk/Payload-SDK/blob/159708e6ef8920f41dce11ce666a3446e91481f5/samples/sample_c/module_sample/flight_control/test_flight_control.c#L1361)

In short we effectively only want to set either a yaw angle or yaw rate and let the mission continue as normal.

We have also posted the details of our problem here: [https://stackoverflow.com/questions/79028784/dji-payload-sdk-control-yaw-independently-while-in-a-waypoint-mission](https://stackoverflow.com/questions/79028784/dji-payload-sdk-control-yaw-independently-while-in-a-waypoint-mission)

dji-dev, 2024-09-27:
Agent comment from Leon in <a href='https://djisdksupport.zendesk.com/agent/tickets/118297'>Zendesk ticket #118297</a>:<br><br>Hello, unfortunately, PSDK does not support breakpoint flight. Your adjustment can only be completed using the flight control function. That is to say, after reaching a certain position, you need to stop the waypoint mission first, then adjust the position, and then re-upload and start the waypoint mission. This requires you to store the remaining unfinished waypoint mission information before adjustment, and then re-upload this information to the drone after adjusting the yaw angle, and then re-upload the waypoint mission starting from the current position.
<br><p><em><small>°°°</small></em></p>

Ellislee1, 2024-09-30:
@dji-dev What would happen if the gimbal has a software angle limit, say of `0-1 (deg)` but is commanded to make a turn greater than this?
Would this yaw the aircraft to compensate?

dji-dev, 2024-10-08:
Agent comment from Leon in <a href='https://djisdksupport.zendesk.com/agent/tickets/118297'>Zendesk ticket #118297</a>:<br><br>Hello, the aircraft will not be moved. The control of the gimbal is independent of the aircraft. The control of the gimbal will not affect the aircraft. If the limit is exceeded, an error will be reported. Before executing the parameters you entered, the software will check whether the constraints are met.
<br><p><em><small>°°°</small></em></p>

closed 2024-10-08 by Ellislee1