dji-sdk/Payload-SDK #210: MAVic 3E ethernet - OTG

sehadnassim, 2024-10-11:
I have the following setup 
[image]
E-Port --> Devkit (host mode) ---> USB C OTG ---> USB network card --> Ethernet of  Raspberry Pi CM4 Io bord 

lsusb does not show the usb network card. and here is my log after testing the Psdk : 

6.460][linker]-[Warn]-[DjiProtocol_v1Unpack:187) protocol frame crc16 error, exp:0x3F2A cur:0x0000 
[6.504][linker]-[Warn]-[DjiProtocol_v1Unpack:187) protocol frame crc16 error, exp:0x9535 cur:0x0040 
[6.520][linker]-[Warn]-[DjiProtocol_v1Unpack:187) protocol frame crc16 error, exp:0x4463 cur:0x0000 
[6.540][linker]-[Warn]-[DjiProtocol_v1Unpack:187) protocol frame crc16 error, exp:0x19D5 cur:0x706F 
[6.580][linker]-[Warn]-[DjiProtocol_v1Unpack:187) protocol frame crc16 error, exp:0x031B cur:0x0000 
[6.605][linker]-[Warn]-[DjiProtocol_v1Unpack:187) protocol frame crc16 error, exp:0x9C0B cur:0x0040 
[6.666][linker]-[Warn]-[DjiProtocol_v1Unpack:187) protocol frame crc16 error, exp:0x0692 cur:0x0000 
[6.704][linker]-[Warn]-[DjiProtocol_v1Unpack:187) protocol frame crc16 error, exp:0x7EC2 cur:0x0040 
[6.720][linker]-[Warn]-[DjiProtocol_v1Unpack:187) protocol frame crc16 error, exp:0xD524 cur:0x00C0 
[6.763][linker]-[Warn]-[DjiProtocol_v1Unpack:187) protocol frame crc16 error, exp:0x9F67 cur:0x0000 
[6.822][linker]-[Warn]-[DjiProtocol_v1Unpack:187) protocol frame crc16 error, exp:0x0AFC cur:0x00C0 
[6.841][linker]-[Warn]-[DjiProtocol_v1Unpack:187) protocol frame crc16 error, exp:0x8A85 cur:0x5F63 
[6.884][linker]-[Warn]-[DjiProtocol_v1Unpack:187) protocol frame crc16 error, exp:0xD466 cur:0x0000 
[6.942][linker]-[Warn]-[DjiProtocol_v1Unpack:187) protocol frame crc16 error, exp:0x2642 cur:0xCA2F 
[7.005][linker]-[Warn]-[DjiProtocol_v1Unpack:187) protocol frame crc16 error, exp:0x57BB cur:0x0F55 
[7.023][linker]-[Warn]-[DjiProtocol_v1Unpack:187) protocol frame crc16 error, exp:0x38DE cur:0x2200 
[7.028][linker]-[Warn]-[DjiProtocol_v1Unpack:187) protocol frame crc16 error, exp:0x1B9A cur:0x6F00 
[7.062][linker]-[Warn]-[DjiProtocol_v1Unpack:187) protocol frame crc16 error, exp:0x7AED cur:0x0000 
[7.100][adapter]-[Info]-[DjiPayloadNegotiate_Init:228) Waiting payload negotiate finish. 
[7.105][linker]-[Warn]-[DjiProtocol_v1Unpack:187) protocol frame crc16 error, exp:0x473D cur:0x0040 
[7.122][linker]-[Warn]-[DjiProtocol_v1Unpack:187) protocol frame crc16 error, exp:0xC45C cur:0x2140 
[7.145][linker]-[Warn]-[DjiProtocol_v1Unpack:187) protocol frame crc16 error, exp:0x6429 cur:0x70CA 
[7.204][linker]-[Warn]-[DjiProtocol_v1Unpack:187) protocol frame crc16 error, exp:0x2BE1 cur:0x0F55 
[7.223][linker]-[Warn]-[DjiProtocol_v1Unpack:187) protocol frame crc16 error, exp:0x0059 cur:0x2200 
[7.247][linker]-[Warn]-[DjiProtocol_v1Unpack:187) protocol frame crc16 error, exp:0x7119 cur:0x0000 
[7.286][linker]-[Warn]-[DjiProtocol_v1Unpack:187) protocol frame crc16 error, exp:0x1918 cur:0x0000 
[7.323][linker]-[Warn]-[DjiProtocol_v1Unpack:187) protocol frame crc16 error, exp:0xE91E cur:0xDA04 
[7.368][linker]-[Warn]-[DjiProtocol_v1Unpack:187) protocol frame crc16 error, exp:0x353F cur:0x5503 
[7.393][linker]-[Warn]-[DjiProtocol_v1Unpack:187) protocol frame crc16 error, exp:0x368B cur:0x0000 
[7.451][linker]-[Warn]-[DjiProtocol_v1Unpack:187) protocol frame crc16 error, exp:0xBE37 cur:0x0000 
[7.487][linker]-[Warn]-[DjiProtocol_v1Unpack:187) protocol frame crc16 error, exp:0x417A cur:0x0000 
[7.522][linker]-[Warn]-[DjiProtocol_v1Unpack:187) protocol frame crc16 error, exp:0x2868 cur:0x2200 
[7.548][linker]-[Warn]-[DjiProtocol_v1Unpack:187) protocol frame crc16 error, exp:0x1E87 cur:0x86CA 
[7.588][linker]-[Warn]-[DjiProtocol_v1Unpack:187) protocol frame crc16 error, exp:0x7B61 cur:0x0000 
[7.623][linker]-[Warn]-[DjiProtocol_v1Unpack:187) protocol frame crc16 error, exp:0x13EF cur:0x2200 
[7.673][linker]-[Warn]-[DjiProtocol_v1Unpack:187) protocol frame crc16 error, exp:0xC6F7 cur:0x0000 
[7.707][linker]-[Warn]-[DjiProtocol_v1Unpack:187) protocol frame crc16 error, exp:0x9E4A cur:0x0040 
[7.722][linker]-[Warn]-[DjiProtocol_v1Unpack:187) protocol frame crc16 error, exp:0xE279 cur:0x2200 
[7.850][linker]-[Warn]-[DjiProtocol_v1Unpack:187) protocol frame crc16 error, exp:0xB18E cur:0x0000 
[7.888][linker]-[Warn]-[DjiProtocol_v1Unpack:187) protocol frame crc16 error, exp:0xE48A cur:0x0000 
[7.950][linker]-[Warn]-[DjiProtocol_v1Unpack:187) protocol frame crc16 error, exp:0x5E44 cur:0x0000 
[7.988][linker]-[Warn]-[DjiProtocol_v1Unpack:187) protocol frame crc16 error, exp:0x24DE cur:0x0000 
[8.023][linker]-[Warn]-[DjiProtocol_v1Unpack:187) protocol frame crc16 error, exp:0x7EAC cur:0xDA04 
[8.033][linker]-[Warn]-[DjiProtocol_v1Unpack:187) protocol frame crc16 error, exp:0xFCD6 cur:0x2FC6 
[8.072][linker]-[Warn]-[DjiProtocol_v1Unpack:187) protocol frame crc16 error, exp:0x3BB4 cur:0x0000 
[8.100][adapter]-[Error]-[DjiAccessAdapter_Init:381) Payload negotiate error, returnCode = 225 
[8.100][core]-[Error]-[DjiCore_Init:126) Access adapter init error, stat:225 
[8.104][linker]-[Warn]-[DjiProtocol_v1Unpack:187) protocol frame crc16 error, exp:0xDFCF cur:0x0040 
[8.122][linker]-[Warn]-[DjiProtocol_v1Unpack:187) protocol frame crc16 error, exp:0x6E8B cur:0x3C55 
[8.169][linker]-[Warn]-[DjiProtocol_v1Unpack:187) protocol frame crc16 error, exp:0x548F cur:0x55DA 
[8.196][linker]-[Warn]-[DjiProtocol_v1Unpack:187) protocol frame crc16 error, exp:0x9533 cur:0x0300 
[8.257][linker]-[Warn]-[DjiProtocol_v1Unpack:187) protocol frame crc16 error, exp:0xCBFD cur:0x0000 
[8.293][linker]-[Warn]-[DjiProtocol_v1Unpack:187) protocol frame crc16 error, exp:0x1E77 cur:0x0000 
[8.323][linker]-[Warn]-[DjiProtocol_v1Unpack:187) protocol frame crc16 error, exp:0x38B5 cur:0x2200 
[8.395][linker]-[Warn]-[DjiProtocol_v1Unpack:187) protocol frame crc16 error, exp:0xD046 cur:0x0000 
[8.413][linker]-[Warn]-[DjiProtocol_v1Unpack:187) protocol frame crc16 error, exp:0xB5F5 cur:0x6F74 
[8.480][linker]-[Warn]-[DjiProtocol_v1Unpack:187) protocol frame crc16 error, exp:0x9CC4 cur:0x0000 
[8.523][linker]-[Warn]-[DjiProtocol_v1Unpack:187) protocol frame crc16 error, exp:0xF95A cur:0x2200 
[8.577][linker]-[Warn]-[DjiProtocol_v1Unpack:187) protocol frame crc16 error, exp:0x81A5 cur:0x0000 
[8.623][linker]-[Warn]-[DjiProtocol_v1Unpack:187) protocol frame crc16 error, exp:0xB5D7 cur:0x2200 
[8.660][linker]-[Warn]-[DjiProtocol_v1Unpack:187) protocol frame crc16 error, exp:0xBC13 cur:0x044A 
[8.693][linker]-[Warn]-[DjiProtocol_v1Unpack:187) protocol frame crc16 error, exp:0xC81C cur:0x0000 
[8.722][linker]-[Warn]-[DjiProtocol_v1Unpack:187) protocol frame crc16 error, exp:0x89A9 cur:0x000F 
[8.775][linker]-[Warn]-[DjiProtocol_v1Unpack:187) protocol frame crc16 error, exp:0x4100 cur:0x554F 
[8.806][linker]-[Warn]-[DjiProtocol_v1Unpack:187) protocol frame crc16 error, exp:0x1055 cur:0x0040 
[8.878][linker]-[Warn]-[DjiProtocol_v1Unpack:187) protocol frame crc16 error, exp:0xAC66 cur:0x0000 
[8.904][linker]-[Warn]-[DjiProtocol_v1Unpack:187) protocol frame crc16 error, exp:0x6BBC cur:0x040F 
[8.960][linker]-[Warn]-[DjiProtocol_v1Unpack:187) protocol frame crc16 error, exp:0x18A7 cur:0x0000 
[9.002][linker]-[Warn]-[DjiProtocol_v1Unpack:187) protocol frame crc16 error, exp:0xCE78 cur:0x0000 
[9.023][linker]-[Warn]-[DjiProtocol_v1Unpack:187) protocol frame crc16 error, exp:0x2731 cur:0x2200 
[9.065][linker]-[Warn]-[DjiProtocol_v1Unpack:187) protocol frame crc16 error, exp:0x2F36 cur:0x0000 
terminate called after throwing an instance of 'std::runtime_error'
  what():  Core init error.
Aborted 

I have tested alot of things now for a whole week and nothing works.

dji-dev, 2024-10-11:
Agent comment from Leon in <a href='https://djisdksupport.zendesk.com/agent/tickets/119118'>Zendesk ticket #119118</a>:<br><br>Hello, regarding the hardware connection, you can refer to here. We do not support all USB network card models, only AX88179A and RTL8152. In addition, you can attach a complete debug log. So that we can better support you.
PSDK connection of various aircraft models
<br><p><em><small>°°°</small></em></p>

sehadnassim, 2024-10-11:
I am using RTL8152 with OTG cable to E-Port dev kit. see new setup : 

[image]

Network card adapter : 

[image]

I still get the error. Here is the log attached 
[DJI_0000_20241009_08-55-54.log](https://github.com/user-attachments/files/17342768/DJI_0000_20241009_08-55-54.log)

dji-dev, 2024-10-12:
Agent comment from Leon in <a href='https://djisdksupport.zendesk.com/agent/tickets/119118'>Zendesk ticket #119118</a>:<br><br>Thank you for your patience. We have reviewed the logs you attached, and the main problem here is that the serial port verification failed, indicating that there is a disconnection between your PSDK program and the aircraft. In this regard, we recommend changing the USB-TTL (FT232 model is recommended) used for serial port connection. You can also try to adjust the baud rate, for example, lower it.
<br><p><em><small>°°°</small></em></p>

sehadnassim, 2024-10-14:
Indeed the issue was on the FTDI.