dji-sdk/Payload-SDK #211: Mavic 3E stream ffmpeg

sehadnassim, 2024-10-14:
Is it possible using the PSDK to retrieve the video stream from the DJI camera and play it on the platform using fflpay or stream it to an RTMP server using ffmpeg ?

dji-dev, 2024-10-14:
Agent comment from Leon in <a href='https://djisdksupport.zendesk.com/agent/tickets/119296'>Zendesk ticket #119296</a>:<br><br>Hello, you can use PSDK to obtain the video stream data of the drone. They are video streams in H264 format. You can push them twice or do other processing on them.
<br><p><em><small>°°°</small></em></p>

sehadnassim, 2024-10-14:
Any example of how to play the stream using ffplay. I currently have the following code but it says address already in use 

/**
 ********************************************************************
 * @file    main.cpp
 * @brief
 *
 * @copyright (c) 2021 DJI. All rights reserved.
 *
 * All information contained herein is, and remains, the property of DJI.
 * The intellectual and technical concepts contained herein are proprietary
 * to DJI and may be covered by U.S. and foreign patents, patents in process,
 * and protected by trade secret or copyright law.  Dissemination of this
 * information, including but not limited to data and other proprietary
 * material(s) incorporated within the information, in any form, is strictly
 * prohibited without the express written consent of DJI.
 *
 * If you receive this source code without DJI’s authorization, you may not
 * further disseminate the information, and you must immediately remove the
 * source code and notify DJI of its removal. DJI reserves the right to pursue
 * legal actions against you for any loss(es) or damage(s) caused by your
 * failure to do so.
 *
 *********************************************************************
 */

/* Includes ------------------------------------------------------------------*/
#include <liveview/test_liveview_entry.hpp>
#include <perception/test_perception_entry.hpp>
#include <flight_control/test_flight_control.h>
#include <gimbal/test_gimbal_entry.hpp>
#include <hms/test_hms.h>
#include <waypoint_v2/test_waypoint_v2.h>
#include <waypoint_v3/test_waypoint_v3.h>
#include <gimbal_manager/test_gimbal_manager.h>
#include "application.hpp"
#include "fc_subscription/test_fc_subscription.h"
#include <gimbal_emu/test_payload_gimbal_emu.h>
#include <camera_emu/test_payload_cam_emu_media.h>
#include <camera_emu/test_payload_cam_emu_base.h>
#include <dji_logger.h>
#include "widget/test_widget.h"
#include "widget/test_widget_speaker.h"
#include <power_management/test_power_management.h>
#include "data_transmission/test_data_transmission.h"
#include <camera_manager/test_camera_manager.h>
#include "camera_manager/test_camera_manager_entry.h"
#include "dji_liveview.h" // Ensure this includes the necessary types and functions

#include <iostream>
#include <cstdio>
#include <cstdlib>
#include <unistd.h>
#include <string>
#include <thread>
#include <chrono>
#include <sys/types.h>  // for socket types
#include <sys/socket.h> // for socket(), connect(), sendto()
#include <netinet/in.h> // for sockaddr_in
#include <arpa/inet.h>  // for inet_addr()
#include <unistd.h>     // for close()

// Function declarations
static void StartFFplayStream(const std::string& streamUrl);
static void DjiTest_FpvCameraStreamCallback(E_DjiLiveViewCameraPosition position, const uint8_t* data, uint32_t dataSize);

// Main function to run the application
int main(int argc, char **argv)
{
    Application application(argc, argv);
    T_DjiReturnCode returnCode;
    
    // Initialize the live view
    returnCode = DjiLiveview_Init();
    if (returnCode != DJI_ERROR_SYSTEM_MODULE_CODE_SUCCESS) {
        std::cerr << "Liveview init failed, error code: 0x" << std::hex << returnCode << std::endl;
        return returnCode;
    }

    // Start the FPV camera stream
    returnCode = DjiLiveview_StartH264Stream(DJI_LIVEVIEW_CAMERA_POSITION_FPV, 
                                             DJI_LIVEVIEW_CAMERA_SOURCE_DEFAULT,
                                             DjiTest_FpvCameraStreamCallback);
    if (returnCode != DJI_ERROR_SYSTEM_MODULE_CODE_SUCCESS) {
        std::cerr << "Request h264 of FPV failed, error code: 0x" << std::hex << returnCode << std::endl;
        return returnCode;
    }

    // Start ffplay stream
    StartFFplayStream("udp://192.168.90.2:30082"); // Modify the address as needed

    // Main loop or any other logic here
    while (true) {
        // Keep the application running
        std::this_thread::sleep_for(std::chrono::seconds(1));
    }

    // Stop the stream and deinitialize live view before exiting
    DjiLiveview_StopH264Stream(DJI_LIVEVIEW_CAMERA_POSITION_FPV, DJI_LIVEVIEW_CAMERA_SOURCE_DEFAULT);
    DjiLiveview_Deinit();
    return 0;
}

// Callback function to handle the FPV camera stream
static void DjiTest_FpvCameraStreamCallback(E_DjiLiveViewCameraPosition position, const uint8_t* data, uint32_t dataSize) {
    // Write data to a UDP socket to stream to ffplay
    int sock = socket(AF_INET, SOCK_DGRAM, 0);
    if (sock < 0) {
        std::cerr << "Failed to create socket!" << std::endl;
        return;
    }

    struct sockaddr_in destAddr;
    destAddr.sin_family = AF_INET;
    destAddr.sin_port = htons(30082); // Port for ffplay
    destAddr.sin_addr.s_addr = inet_addr("127.0.0.1");

    ssize_t sentBytes = sendto(sock, data, dataSize, 0, (struct sockaddr*)&destAddr, sizeof(destAddr));
    if (sentBytes < 0) {
        std::cerr << "Failed to send data!" << std::endl;
    }

    close(sock); // Close the socket
}

static void StartFFplayStream(const std::string& streamUrl) {
    std::string command = "export XDG_RUNTIME_DIR=/run/user/" + std::to_string(getuid()) + " && ffplay " + streamUrl + " &";
    system(command.c_str()); // Launch ffplay as a subprocess
}

dji-dev, 2024-10-15:
Agent comment from Leon in <a href='https://djisdksupport.zendesk.com/agent/tickets/119296'>Zendesk ticket #119296</a>:<br><br>Hello, if you want to see the video stream directly on the terminal, you can refer to the processing in C++sample: payload-sdk\samples\sample_c++\module_sample\liveview\test_liveview.cpp
<br><p><em><small>°°°</small></em></p>

sehadnassim, 2024-10-17:
The example code for live display does not do anything in MAVIC 3E : 
[image]

dji-dev, 2024-10-18:
Agent comment from Leon in <a href='https://djisdksupport.zendesk.com/agent/tickets/119296'>Zendesk ticket #119296</a>:<br><br>Hello, you can check your ffmpeg and opencv versions. We have officially recommended versions. In addition, you need to confirm whether the libusb component is successfully installed. This component will also affect the normal use of this function. Finally, it is recommended that you use the code of the manifold2 project. The codes of other projects may not be updated in time. In order to avoid these effects, it is recommended that you use the code of this project.
https://developer.dji.com/doc/payload-sdk-tutorial/en/quick-start/config-develop-environment.html
<br><p><em><small>°°°</small></em></p>