dji-sdk/Payload-SDK #212: PSDK Virtual Joystick Commands During Missions

Ellislee1, 2024-10-15:
Is we have a mission active on the drone activated via the Cloud SDK and had some code on the drone (PSDK) using a virtual Joystick:
Can we issue yaw commands via the virtual joystick such that it does not stop the mission
```C
 T_DjiFlightControllerJoystickCommand joystickCommand = {positionCommand.x, positionCommand.y, offsetDesired.z + originHeightBaseHomePoint, yawDesiredInDeg};
        DjiFlightController_ExecuteJoystickAction(joystickCommand);
```
This Idea comes from the fact that, in flight, we can yaw the aircraft with the controller during a mission. 
Our concern comes from the the line above where it appears we have to pass it x,y,z as well as yaw.

All we want to know is can we send a yaw virtual joystick command without it interrupting the mission and if so what is the method for this.

Ellislee1, 2024-10-16:
The mission is activated either dronehub2 or via the pilot2 app.

dji-dev, 2024-10-17:
Agent comment from Leon in <a href='https://djisdksupport.zendesk.com/agent/tickets/119501'>Zendesk ticket #119501</a>:<br><br>Hello, no matter which end is activated from, the status will be stored in the flight control system. If PSDK intervenes in the control during this period, the flight control will have judgment logic to prevent this behavior, unless the current task is paused or stopped directly, at which time PSDK can take over or control the drone.
<br><p><em><small>°°°</small></em></p>

closed 2024-10-17 by Ellislee1