dji-sdk/Payload-SDK #214: DJI FREERTOS STM32F407 Issue setting up and error codes

Ellislee1, 2024-10-21:
We are trying to set up the PSDK to run on an STM32 - The documentation for which is severely lacking.
We have finally got it to compile but we get within the `DJI_CoreInit` function an error code `225`. We have put this in a loop and after a while this code changes to `236`.

What do these error codes mean and what is the fix?

dji-dev, 2024-10-23:
Agent comment from Leon in <a href='https://djisdksupport.zendesk.com/agent/tickets/119907'>Zendesk ticket #119907</a>:<br><br>Hello, both 225 and 236 have Core initialization failure (core program initialization failure). Which model are you using? We have connection guidelines for different models, I hope it will be helpful to you. You can follow the connection here first.
<br><p><em><small>°°°</small></em></p>

Ellislee1, 2024-10-29:
We are using the M3D

dji-dev, 2024-10-30:
Agent comment from Leon in <a href='https://djisdksupport.zendesk.com/agent/tickets/119907'>Zendesk ticket #119907</a>:<br><br>Hello, you can refer to here for its connection method; the document has a network disk link, which contains a connection video about RTOS.
PSDK connection of various aircraft models
<br><p><em><small>°°°</small></em></p>

Ellislee1, 2024-10-30:
Sorry there was no link...

dji-dev, 2024-10-30:
Agent comment from Leon in <a href='https://djisdksupport.zendesk.com/agent/tickets/119907'>Zendesk ticket #119907</a>:<br><br>https://sdk-forum.dji.net/hc/en-us/articles/34834641217305-PSDK-connection-of-various-aircraft-models
<br><p><em><small>°°°</small></em></p>